RTXI/rtxi #52: update(REFRESH) for DefaultGUI

anselg, 2015-02-03:
I keep overloading the refresh() function in DefaultGUI. By "overloading", I mean copy/pasting what's already in DefaultGUI::refresh() and adding my own stuff below it. It would make more sense if there was an update(REFRESH) built into DefaultGUI, like how there's an update(PAUSE), update(EXIT), etc. 

I'll add the feature now. Just making note of my intentions here (and spamming your inboxes while I'm at it).

yapatel, 2015-02-03:
can you point me to an example code snippet of your’s so I can see what you mean by “copy/pasting”...

> I keep overloading the refresh() function in DefaultGUI. By "overloading", I mean copy/pasting what's already in DefaultGUI::refresh() and adding my own stuff below it. It would make more sense if [...]

Yogi A. Patel
@yapatel525 <https://twitter.com/yapatel525>

anselg, 2015-02-03:
[membrane-test](https://github.com/RTXI/membrane-test/blob/master/membrane-test.cpp#L589)

I copied the stuff in DefaultGUI and added my bit to the bottom. 

Some of our modules, when they do computationally intensive things that don't need to be (or can't be) done in real-time, do them in the GUI thread. Others need to update displays and messages. 

The 1.4 versions of them have timers that set when to update the GUI, and when I rebuilt them to abstract from DefaultGUI, I overloaded refresh to have the same timer trigger all the GUI changes. 

[A module with a second timer (refresh not overloaded. rather, two refresh functions made)](https://github.com/RTXI/spike-triggered-average/blob/master/spike-triggered-average.cpp#L122)

anselg, 2015-02-04:
I'll hold off on committing any changes, then. 

As far as adding code is concerned, it's a matter of adding REFRESH to update_flags_t and adding update(REFRESH) to the end of refresh(void) {...}. 
````
enum update_flags_t {
      INIT,     /*!< The parameters need to be initialized.         */
      MODIFY,   /*!< The parameters have been modified by the user. */
      PERIOD,   /*!< The system period has changed.                 */
      PAUSE,    /*!< The Pause button has been activated            */
      UNPAUSE,  /*!< When the pause button has been deactivated     */
      EXIT,     /*!< When the module has been told to exit        */
      REFRESH,  /*!< When the screen refreshes */
};
````

````
void DefaultGUIModel::refresh(void) {
   for (std::map<QString, param_t>::iterator i = parameter.begin(); i!= parameter.end(); ++i) {
      if (i->second.type & (STATE | EVENT)) {
         i->second.edit->setText(QString::number(getValue(i->second.type, i->second.index)));
         palette.setBrush(i->second.edit->foregroundRole(), Qt::darkGray);
         i->second.edit->setPalette(palette);
      } else if ((i->second.type & PARAMETER) && !i->second.edit->isModified()
            && i->second.edit->text() != *i->second.str_value) {
         i->second.edit->setText(*i->second.str_value);
      } else if ((i->second.type & COMMENT) && !i->second.edit->isModified()
            && i->second.edit->text() != QString::fromStdString(getValueString(COMMENT, i->second.index))) {
         i->second.edit->setText(QString::fromStdString(getValueString(COMMENT, i->second.index)));
      }
   }
   pauseButton->setChecked(!getActive());

   update(REFRESH);
}
````

closed 2015-06-17 by yapatel