RTXI/rtxi #53: maximum downsampling rate for the data recorder

anselg, 2015-03-04:
https://github.com/RTXI/rtxi/blob/qt4/plugins/data_recorder/data_recorder.cpp#L517

Why is the maximum downsampling rate 2?

yapatel, 2015-03-05:
This is one of those “subjective” decisions…not many downsample by more than a factor of two. What rate are you aiming for/wanting?

> 
> https://github.com/RTXI/rtxi/blob/qt4/plugins/data_recorder/data_recorder.cpp#L517 <https://github.com/RTXI/rtxi/blob/qt4/plugins/data_recorder/data_recorder.cpp#L517>
> Why is the maximum downsampling rate 2?

Yogi A. Patel
@yapatel525 <https://twitter.com/yapatel525>

anselg, 2015-03-05:
When I record the RT benchmarks module for a couple minutes at 50 kHz, the files get massive. When I plot the data with R and ggplot, their voraciousness for memory makes the issue worse, and my system starts thrashing. I want to downsample it by, say, 100-200. Enough for me to record for 30 min. to an hour without the *.h5 files becoming unmanageably large. 

I suppose that 100-200 is a bad idea when collecting experimental data, but I don't think that RTXI should block it. The 1.4 version lets you downsample between 1-1000.

How about we set the max to 500?

closed 2015-03-08 by yapatel