RTXI/rtxi #55: Data Recording Problems

El-Jefe, 2015-03-25:
In the case where I have 2 digital inputs, the data recorder does not write either to the designated file.

anselg, 2015-04-02:
I've sometimes had issues with the data recorder, but I don't have a method of deterministically reproducing them. I'll just list what I've encountered on what machine: 

Optiplex 9020  
---------
1. Open the data recorder and then close it. RTXI sometimes freezes, and I have to restart it. 
2. Open the data recorder and open an existing *.h5 files. The filename doesn't appear in the field, and when I close the module, RTXI freezes. This may be the same error as 1. 
3. Hit the `Start Recording` button. Sometimes it doesn't record. None of the button states change, RTXI freezes, and the terminal is released. 
4. Hit `Start Recording`. RTXI just flat-out crashes. There's also an error message:
````
QObject::connect: Cannot queue arguments of type 'Qt::WindowStates'
(Make sure 'Qt::WindowStates' is registered using qRegisterMetaType().)
````

yapatel, 2015-04-21:
These were due to improper initialization of the new thread and it's child process.

anselg, 2015-04-21:
Using commit 7a85b28b8e793bca77a3344f47d2e9250d6e00e5

I'm still seeing problems with the Data Recorder. RTXI sometimes still freezes when I hit "Start Record" and once when I opened an *.h5 file. (This is all when I'm trying to record the Performance Measurement module.) 

I still haven't found a way of reproducing this deterministically. None of the freezes produce terminal output. 

I'd also include a backtrace from GDB, but when RTXI freezes, I can't do anything within GDB. `CTRL+c` produces no response, and running `sudo pkill rtxi` or even `sudo pkill gdb` doesn't do anything. Running `sudo pkill gnome-terminal` from another tty will kill GDB and leave a zombie process in its place. (Not that killing the terminal does any help in debugging.)

yapatel, 2015-04-21:
I was made aware of this problem earlier and have found the source. I'm working on it. It has to do with the RT Benchmarks module. There appears to have been bad code inserted into the module.

yapatel, 2015-04-21:
Correction: it has to do with the data recorder, specifically an object cast that is used to write to HDF5.

There is an object that is not properly initialized somewhere. Depending on the steps you take before to using the data recorder, that object may get initialized - hence why this error is hard to reproduce.

closed 2015-04-26 by yapatel