RTXI/rtxi #64: Data Recorder UI does not respond to Sync module

anselg, 2015-04-21:
When the sync module is started with the data recorder option checked, the data will get recorded (you can see the Trial number increment every time you unpause Sync), but the Data Recorder UI will not change state. (The start button remains enabled, etc.)

yapatel, 2015-04-21:
Sorry - forgot to link to the new data recorder events/states.

anselg, 2015-04-22:
Using rtxi: commit 6e824c2de0c8dfeccc96729062fe172b6de34650
Using sync: https://github.com/RTXI/sync/commit/e690c9a54430b65c88f6035d092a1e159e3a0170

I'm still getting the same error. The data recorder UI doesn't update when the sync module is activated, but it still records. 

I also noticed that `rtxi/deps/rtxi_includes/data_recorder.h` is removed. The sync module uses the `DataRecorder::(start|stop)Recording` functions, and it won't compile without that header file. Maybe instead of two copies of `data_recorder.h`, one could just be a simlink. 

I'm getting the bug after copying the current version of `data_recorder.h` (per the RTXI commit above) to `/usr/local/lib/rtxi_includes/` and then re-compiling and installing the sync module.

yapatel, 2015-04-22:
When source files at the base of RTXI are modified, the install scripts should be used to reinstall RTXI properly.

Do the following:

in your root rtxi directory, run "sudo make clean; sudo make uninstall"

then, just for kicks sake - run "git pull" to make sure your source is updated on the qt4 branch

then, cd into the scripts folder and run "bash install_dependencies.sh" and then afterwards run "bash install_rtxi.sh"

After reinstalling RTXI, run "sudo make clean" in the sync directory (with the latest source), run “make" and "sudo make install".

This will take care of the problem you are facing.

> 
> Using rtxi: commit 6e824c2 <https://github.com/RTXI/rtxi/commit/6e824c2de0c8dfeccc96729062fe172b6de34650>
> Using sync: RTXI/sync@e690c9a <https://github.com/RTXI/sync/commit/e690c9a54430b65c88f6035d092a1e159e3a0170>
> I'm still getting the same error. The data recorder UI doesn't update when the sync module is activated, but it still records.
> 
> I also noticed that rtxi/deps/rtxi_includes/data_recorder.h is removed. The sync module uses the DataRecorder::(start|stop)Recording functions, and it won't compile without that header file. Maybe instead of two copies of data_recorder.h, one could just be a simlink.
> 
> I'm getting the bug after copying the current version of data_recorder.h (per the RTXI commit above) to /usr/local/lib/rtxi_includes/ and then re-compiling and installing the sync module.

Yogi A. Patel
@yapatel525 <https://twitter.com/yapatel525>

closed 2015-04-26 by yapatel