RTXI/rtxi #68: Oscilloscope UI doesn't update properly when plotting input channels for modules

anselg, 2015-05-26:
When plotting the inputs to modules (neuron in this case), the oscilloscope UI will revert back to the 'disabled' appearance when "Apply" is hit to enable a channel, but it will still plot the signal like it's supposed to. 

See the below screenshot. The signal-generator is connected to the neuron module. The signal waveform is in red, and Iapp for the neuron module in blue. Notice that the UI for Iapp has buttons greyed out, as if the Iapp channel is disabled in the oscilloscope. 

[image]

This behaviour changes when the the Vm output of the neuron module is enabled. Instead of the oscilloscope showing neuon's Iapp being disabled, it is enabled, even if it's not being plotted. See below:
[image]

It seems the UI is looking at the state of the Output channel of the module to determine the state of the Input channel.

yapatel, 2015-05-27:
I am unable to reproduce this error; please pull the latest master branch, recompile, and verify that you are using the latest version Help->About RTXI (should show rc3).

anselg, 2015-05-27:
Fixes to the issue are already on the master branch. Pulling them into the qt5 branch fixed the issue.

closed 2015-05-27 by anselg