RTXI/rtxi #72: Digital I/O does not work properly with the analogy driver.

anselg, 2015-06-12:
This statement isn't very well characterized on my part. It's more of an assumption based on troubleshooting several things. 

**What I'm trying to do:** Use the digital lines to send a command signal to an amplifier. 

**What I've tested:**  
 - RTXI 1.4: I can successfully send the command signals via digial I/O using the comedi and NI drivers, but not the analogy driver. The comedi driver and NI driver were used as-is from the v1.4 source code (`comedi_device` and `ni_device`), and the analogy driver is backported from the version currently at the head of the master branch. (By "backported", I mean I changed qt4-specific string commands to be qt3-compatible.)  
 - RTXI 2.0 - I cannot send command signals using the analogy driver. I have not tried this with the NI driver, and I could not for the life of me get `comedi` to compile with RTAI 4.1 (I last tried a few months ago).  

The module that send the command signal is `am-amp2400`. When using RTXI 1.4, I used the code in the v1.4 branch, and when using RTXI 2.0, I used the code on that module's master branch. 

What the command signal does is tell the amplifier what mode to use. When the signal doesn't work (i.e., when the analogy driver is in use), the amplifier does not change mode properly. Instead, the amplifier produces output inconsistent with expected values. 

Given that I can get digital I/O working in v1.4 using comedi and ni driver and that essentially the same analogy driver will not work in v1.4 or v2.0, the likeliest conclusion is that there is some error with the way the analogy driver (probably something in the `analogy_device` files) is handling the digital signals.

yapatel, 2015-06-13:
I've tested the DIO on multiple computers and it does work with the analogy drivers. I have approximately 6 DIO in use and actively working. 

Let's start smaller - using the TTL module, set up the output to a DIO line and check that the scope shows an output?

and what is the input signal that the amplifier is looking for? Is it looking for a single rising edge?

anselg, 2015-06-15:
The amp takes three inputs that are set to either 0 or 5 V. The combination
of inputs that are 'off' and 'on' tells the amp which mode to use. I'll
look through the manual to see how it differentiates between 0 and 5 V.

I don't think anyone is using the rig today. I'll update you as soon as
I've tried those simple IO tests.

yapatel, 2015-06-24:
This should now be resolved. The issue was that the bit masks for the DIO were not being updated properly (errors in | and ^ operations). Please checkout the [diofix] (https://github.com/RTXI/rtxi/tree/diofix) branch and test your amplifier module.

anselg, 2015-06-24:
The DIO fix works. The amplifier gets the command signal from the analogy
driver.

closed 2015-06-25 by anselg