RTXI/rtxi #76: Trouble with Nvidia graphics card

Wmdhampton, 2015-07-01:
tried installing RTXI from commit: ef4f4c2dbb17c66dae0eb3d961bd1e2d9ca9b414

I am currently having issues using RTXI with my graphics card, in which I am getting the black screen of death after installing RTXI. Upon a cursory glance, I do not believe that there is a nouveau driver for our current graphics card:
Model: GF119 [NVS 310] 
(NVIDIA® Quadro® NVS 310 512MB (2DP) (2DP-DVI adapter)).
Vendor ID: 10de
pci ID: 107d
Instead of installing nouveau, Ubuntu 14.04 installs an nvidia driver. We are thinking about ordering a new graphics card. Does anyone have any recommendations on either how to get our graphics card to work, or what graphics card we should order? The computer that we are using is a Dell Precision Tower 5810. Here is a are a few AMD cards that seem like they should be compatible (though we'd also be happy to hear other suggestions):

AMD FirePro™ W2100 2GB (2 DP) (1 DP to SL-DVI adapter) [add $18.00]
AMD FirePro™ W4100 2GB (4 mDP) (4 mDP-DP adapters) [add $69.75]

yapatel, 2015-07-01:
We do recommend the use of AMD cards. Our tests have shown them to be superior in terms of driver compatibility and performance.

Based upon what I can find about your computer from this [manual](http://topics-cdn.dell.com/pdf/precision-t5810-workstation_Owner's%20Manual_en-us.pdf), the FirePro W4100 should work. It is also listed as being supported by the radeon driver for Ubuntu (

yapatel, 2015-07-02:
A few questions...

1. Which kernel are you running? Since you are getting the black screen of death, you should be able to use "CTRL+ALT+F1". Then, login at the command line and once in, type "uname -r" and copy paste the output here.

2. How did you install rtxi (liveCD or manual)?

Wmdhampton, 2015-07-06:
I was running the Xenomai kernel after a manual installation. When running the Xenomai kernel my cpu took over to run the graphics. We tried installing about 4 different NVidia drivers, but all of them caused the black screen of death. After purchasing the FirePro W4100 everything is working like a charm. Thanks for your help!

yapatel, 2015-07-06:
Great!

closed 2015-07-06 by yapatel