RTXI/rtxi #77: Error when loading newly created module

neuroking, 2015-07-08:
I'm getting an error when I try to load RefLearnNew module in RTXI. This is what it says:

root@nnetoff-OptiPlex-755:/home/RTXI/NetoffModels/RefLearnNew# sudo rtxi

../src/plugin.cpp:51:Plugin::load : failed to load RefLearnNew.so: /usr/local/lib/rtxi/RefLearnNew.so: undefined symbol: _ZN11RefLearnNew16staticMetaObjectE

The compilation process went through without errors. When I try to load this module in RTXI nothing happens, the module doesn't show up and RTXI doesn't crash. I can load other modules. 

Vivek

yapatel, 2015-07-08:
It looks like you have some undefined symbol in your code. Try running:

$c++filt _ZN11RefLearnNew16staticMetaObjectE

It may point you to the location of the variable.

yapatel, 2015-07-08:
When you use custom signals/slots and declare Q_OBJECT in your header file, you have to add moc_<classname>.cpp to the SOURCES list of your makefile.

In your case, your SOURCES list would look like:

SOURCES = 
RefLearnNew.cpp\
moc_RefLearnNew.cpp

closed 2015-07-08 by yapatel