RTXI/rtxi #79: Protocol editor exit button

bqWCMC, 2015-07-09:
In the Clamp Protocol module, in the protocol editor, you can't exit the protocol preview because there's no exit button. When I hit preview again to update the graph because I changed the protocol, RTXI 2.0 crashed. 
Thanks!

anselg, 2015-07-09:
I will add buttons to the preview popup window. I'll try to replicate the crashing issue, too. 

In the meantime, you can close the box by right-clicking the title bar and selecting "Close".

anselg, 2015-07-09:
The close button on the title bar is not displayed on any dialog boxes in RTXI or for any other application. This isn't an RTXI issue. It's design decision of GNOME (the desktop environment). [This is their rationale.](https://blogs.gnome.org/mclasen/2014/01/13/client-side-decorations-continued/) (TLDR - it saves space)

You can close the protocol viewer window via the right-click menu, or by pressing the ESC button. I also added a close button below the plot, too. Download and install the updated code ([instructions here](http://rtxi.org/modules)). 

I haven't been able to crash RTXI 2.0 by opening multiple protocol preview windows. How many did you open before RTXI crashed?

anselg, 2015-07-31:
The issue with the crash was that many, many protocol preview windows were open at once, all of them plotting rather complex protocols. There's a close button in the window now, so there are three ways (by my count) to close the windows: 

1. Close from the right-click menu  
2. Hit `ESC`  
3. Use the new close button.

closed 2015-07-31 by anselg