RTXI/rtxi #84: rtxi benchmarks plotting in R doesn't work.

djlab, 2015-07-31:
I did a new installation of RTXI the last few days, and tried to run the benchmarks.  
/home/dynamic/src/rtxi/scripts/dev/test_rt_kernel.sh runs for 30 minutes, then fails at the end when it tries to plot the data.  The numbers look good as they roll by (latencies of around 1 sec).

Similarly, when I run the "Record the RT benchmarks module" as described, it creates the file,  but ./plot_rtxi_performance.sh FILENAME fails.  I get this error:

dynamic@dynamic:~/src/rtxi/scripts/dev$ ./plot_rtxi_performance.sh ~/Desktop/new_rick_test.h5 
----->Checking for dependencies needed to generate plot.

HDF5 "/home/dynamic/Desktop/new_rick_test.h5" {
DATASET "/Trial1/Synchronous Data/Channel Data" {
   DATATYPE  H5T_ARRAY { [3] H5T_IEEE_F64LE }
   DATASPACE  SIMPLE { ( 86889 ) / ( H5S_UNLIMITED ) }
   DATA {
   }
}
}
[sudo] password for dynamic: 
Loading required package: gridExtra
Loading required package: methods
Error in gtable_table(d, name = "core", fg_params = theme$core$fg_params,  : 
  could not find function "gpar"
Calls: annotation_custom ... <Anonymous> -> do.call -> tableGrob -> gtable_table
Execution halted

The version of R is:
R version 3.0.2 (2013-09-25) -- "Frisbee Sailing"
Copyright (C) 2013 The R Foundation for Statistical Computing
Platform: x86_64-pc-linux-gnu (64-bit) 

I don't use R normally, so I have no idea where to start debugging this, but if it's part of the distribution, it should work.

thanks,
rick

anselg, 2015-07-31:
There are a few things that can be going on here. Could you:  

1. Tell me how you installed RTXI? Did you use the live CD or install scripts?
2. Check if the RTXI repo you're using up-to-date?  You can use `git log` to show where you are in the development tree. 
3. Email me your new_rick_test.h5 file? (ang2049@med.cornell.edu) I'll try to replicate that error on my system.  

One possibility is that there is an issue with the R packages. Another could be the channel names assigned by the data recorder (we changed the way that is done somewhat recently, and it can affect the R scripts if the version you have isn't up-to-date). 

The info I'm asking for should give me a good idea of what's going wrong. Thanks.

yapatel, 2015-07-31:
Due to the fact that a function can not be found, it is most likely that a package is not installed.

Since the dependency installation is built into the script, it is most likely that the package installation failed because something was changed on the repositories.

The process for benchmarking will be changing in the next release, so not a big deal for now. For now, it is easiest to send us your file and we can plot it and send you the plot back.

anselg, 2015-07-31:
The script is running without issue on my machine. (Ignore the top table - the information is from my machine.)  

[image]

Your RTXI repo is up-to-date, so the likely cause is that the packages aren't installed correctly. Try running these commands and then re-trying the `plot_rtxi_performance.sh` script:  
````
sudo apt-get -y install r-base r-cran-ggplot2 r-cran-reshape2 r-cran-hdf5 r-cran-plyr r-cran-scales
wget --no-check-certificate http://cran.r-project.org/src/contrib/gridExtra_0.9.1.tar.gz
tar xf gridExtra_0.9.1.tar.gz
sudo R CMD INSTALL gridExtra
````

anselg, 2015-07-31:
**UPDATE:** Don't run the comments in the last comment. Use these:
````
sudo apt-get -y install r-base r-cran-ggplot2 r-cran-reshape2 r-cran-hdf5 r-cran-plyr r-cran-scales
wget --no-check-certificate http://cran.r-project.org/src/contrib/gridExtra_2.0.0.tar.gz
sudo R CMD INSTALL gridExtra_2.0.0.tar.gz
````

It seems the gridExtra package was updated, and the scripts did not reflect the changes.

djlab, 2015-07-31:
OK, I installed as directed, and it worked.  Well, the system info at the top of the page didn’t show up with the old file, but I’m running it again now.

thanks a bunch for your help, have a great weekend, enjoy the blue moon,
rick

closed 2015-07-31 by yapatel

anselg, 2015-07-31:
Good to hear. Let us know is you have any further issues. If you don't, feel free to ignore the rest of this message. 
_______________

If you do get errors, I have a very good idea as to why. 

Turns out people don't bump version numbers without reason. The 2.0.0 version of gridExtra isn't compatible with the scripts we're using. For RTXI v2.1, we'll be doing performance plots differently, so for now, you can obtain and install the old, compatible, version of gridExtra. 

First, uninstall your current gridExtra. From the terminal, run:  
````
sudo R CMD REMOVE gridExtra
````

Then, install gridExtra v0.9.1:  
````
wget --no-check-certificate https://cran.r-project.org/src/contrib/Archive/gridExtra/gridExtra_0.9.1.tar.gz
sudo R CMD INSTALL gridExtra_0.9.1.tar.gz
````

That should get your system working. On my system, using v2.0.0 causes the error you were originally seeing, and v0.9.1 will produce the expected plot. It seems that you were using v2.0.0 to begin with.

djlab, 2015-08-01:
I did as you suggested, and it works perfectly now.  I ran test_rt_kernel.sh again, for 30 min at 10KHz.  Here’s the output:

rick