RTXI/rtxi #86: can't compile module "dynamic-clamp-file"

djlab, 2015-08-05:
I'm trying to compile the sample module "dynamic-clamp-file" and get errors.  Here's the output of "make":

dynamic@dynamic:~/src/dynamic-clamp-file-master$ make
/usr/local/lib/rtxi/libtool --mode=compile g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c dynamic-clamp-file.cpp -o dynamic-clamp-file.lo
libtool: compile:  g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c dynamic-clamp-file.cpp  -fPIC -DPIC -o .libs/dynamic-clamp-file.o
dynamic-clamp-file.cpp:40:39: fatal error: QtPrintSupport/QPrintDialog: No such file or directory
 #include <QtPrintSupport/QPrintDialog>

When I look in /usr/include/qt4, there are no entries for QtPrintSupport.  They're in Qt5, but all the include-file searches are for qt4.

If I comment out those lines to include files from QtPrintSupport, things progress until I get:
make: *** No rule to make target `/scatterplot.lo', needed by `dynamic_clamp_file.la'.  Stop.

Am I missing some libraries?  Some environment setting? Sorry, I don't normally use Linux at all, so I get frustrated when something doesn't work as it should.

anselg, 2015-08-05:
The scatterplot file should already be installed. Can you check `/usr/local/lib/rtxi_includes/` to see if scatterplot.cpp and its header are there? 

The module will compile on my machine, but its currently using Qt5, not Qt4. I'll get back to in in a little bit once I test the makefile on a qt5 machine.

djlab, 2015-08-05:
Yes, scatterplot.cpp and scatterplot.h are in /usr/local/lib/rtxi_includes

anselg, 2015-08-05:
Okay, then. The module will compile and install on my machine with Qt4 installed, but only with the QPrinter stuff commented out. That's a bug. I'll let you know when that's fixed. My apologies. 

Without the QPrinter stuff, the module will still work, but it won't be able to print screenshots. Have you tried recompiling that module from scratch? Can you run:
````
sudo make clean
make -j`nproc` && sudo make install
````
in your folder for the dynamic_clamp_file module? What this'll do is clear out all the previously compiled files and try to recompile everything from source. 

Also, have you installed another module that had a plotting functionality? 

Thanks.

yapatel, 2015-08-05:
@djlab change the following lines in your dynamic-clamp-file.c:

 34 #include <QtPrintSupport/QPrintDialog>
 35 #include <QtPrintSupport/QPrinter>

to

 34 #include <QPrintDialog>
 35 #include <QPrinter>

Qt5 is not that different than Qt5. The creators simply organized the location of specific files a bit more.

That should let you compile and install the module without losing functionality.

djlab, 2015-08-05:
I still get an error about scatterplot:

dynamic@dynamic:~/src/dynamic-clamp-file-master$ make -j`nproc` && sudo make install
/usr/local/lib/rtxi/libtool --mode=compile g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c dynamic-clamp-file.cpp -o dynamic-clamp-file.lo
libtool: compile:  g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c dynamic-clamp-file.cpp  -fPIC -DPIC -o .libs/dynamic-clamp-file.o
libtool: compile:  g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c dynamic-clamp-file.cpp -o dynamic-clamp-file.o >/dev/null 2>&1
/usr/local/lib/rtxi/libtool --mode=compile g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c moc_dynamic-clamp-file.cpp -o moc_dynamic-clamp-file.lo
libtool: compile:  g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c moc_dynamic-clamp-file.cpp  -fPIC -DPIC -o .libs/moc_dynamic-clamp-file.o
libtool: compile:  g++ -I. -I/usr/local/include/rtxi -pipe -Wall -I. -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/local/lib/rtxi_includes -I/usr/local/lib/qwt/include -I/usr/X11R6/include -I/usr/include/qwt -w -O3 -std=c++11 -c moc_dynamic-clamp-file.cpp -o moc_dynamic-clamp-file.o >/dev/null 2>&1
make: *** No rule to make target `/scatterplot.lo', needed by `dynamic_clamp_file.la'.  Stop.

djlab, 2015-08-05:
lines are truncated after a left-caret, so I can see what you typed.  It's also gone in the email version.  Just leave out the left and right carets and I'd change it.

yapatel, 2015-08-05:
@djlab change the following lines in your dynamic-clamp-file.c:

34 #include QtPrintSupport/QPrintDialog
35 #include QtPrintSupport/QPrinter

to

34 #include QPrintDialog
35 #include QPrinter

That should let you compile and install the module without losing functionality.

anselg, 2015-08-05:
Could you post your makefile? I wonder if it's literally looking for "/scatterplot.lo" at the root of your filesystem.

djlab, 2015-08-05:
Success at compiling, but when I select the compiled plugin from the RTXI menu, nothing happens.

[image]

Here's the Makefile:

dynamic@dynamic:~/src/dynamic-clamp-file-master$ more Makefile 

PLUGIN_NAME = dynamic_clamp_file

RTXI_INCLUDES=/usr/local/lib/rtxi_includes

HEADERS = dynamic-clamp-file.h\
          $(RTXI_INCLUDES)/scatterplot.h\
          $(RTXI_INCLUDES)/incrementalplot.h\
          $(RTXI_INCLUDES)/basicplot.h\
          $(RTXI_INCLUDES)/scrollzoomer.h\
          $(RTXI_INCLUDES)/scrollbar.h\
          $(RTXI_INCLUDES)/plotdialog.h\

SOURCES = dynamic-clamp-file.cpp \
          moc_dynamic-clamp-file.cpp\
          $(RTXI_INCLUDES)/scatterplot.cpp\
          $(RTXI_INCLUDES)/incrementalplot.cpp\
          $(RTXI_INCLUDES)/basicplot.cpp\
          $(RTXI_INCLUDES)/scrollzoomer.cpp\
          $(RTXI_INCLUDES)/scrollbar.cpp\
          $(RTXI_INCLUDES)/plotdialog.cpp\
          $(RTXI_INCLUDES)/moc_scatterplot.cpp\
          $(RTXI_INCLUDES)/moc_incrementalplot.cpp\
          $(RTXI_INCLUDES)/moc_basicplot.cpp\
          $(RTXI_INCLUDES)/moc_scrollzoomer.cpp\
          $(RTXI_INCLUDES)/moc_scrollbar.cpp\
          $(RTXI_INCLUDES)/moc_plotdialog.cpp\

LIBS = -lqwt

### Do not edit below this line ###

include $(shell rtxi_plugin_config --pkgdata-dir)/Makefile.plugin_compile

yapatel, 2015-08-05:
Did you try loading it by going to “Load Plugin” and then selecting the dynamic-clamp-file.so module?

Is there any output on your terminal?

> Success at compiling, but when I select the compiled plugin from the RTXI menu, nothing happens. [...]

Yogi

djlab, 2015-08-05:
Ahh, now it works!  I get the dynamic clamp window.  Rather than go through "Load plugin" and selecting it, I was mistakenly clicking on "dynamo_model_loader.so" (just because I wasn't reading carefully).  My fault.  Sorry.

thanks for your help,
rick

yapatel, 2015-08-05:
Anytime. Thanks for using RTXI for your research, @djlab!

closed 2015-08-05 by yapatel