RTXI/rtxi #87: Compiler Error:../src/rt_os-xenomai.cpp:152:38: error: ‘rt_timer_tsc’ was not declared in this scope

topquant, 2015-08-25:
developer@developer-HP-xw8600-Workstation:~/rtxi-master$ ./configure --enable-xenomai --enable-comedi  --enable-analogy
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for cppunit-config... no
checking for Cppunit - version >= ... no
checking whether make supports nested variables... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for dlopen in -ldl... yes
checking for cos in -lm... yes
checking for cblas_dgemm in -lgslcblas... yes
checking for gsl_blas_dgemm in -lgsl... yes
checking hdf5.h usability... yes
checking hdf5.h presence... yes
checking for hdf5.h... yes
checking for H5Fopen in -lhdf5... yes
checking for H5PTopen in -lhdf5_hl... yes
checking for boostlib >= 1.54... yes
checking rtai-config... rtai-config not found
configure: creating ./config.status
config.status: creating Makefile
config.status: creating rtxi/Makefile
config.status: creating scripts/Makefile
config.status: creating scripts/init_rtxi
config.status: creating scripts/rtxi_comedi
config.status: creating scripts/rtxi_plugin_config
config.status: WARNING:  'scripts/rtxi_plugin_config.in' seems to ignore the --datarootdir setting
config.status: creating plugins/Makefile
config.status: creating plugins/analogy_device/Makefile
config.status: creating plugins/comedi_device/Makefile
config.status: creating plugins/comedilib_device/Makefile
config.status: creating plugins/connector/Makefile
config.status: creating plugins/data_recorder/Makefile
config.status: creating plugins/dynamo_model_loader/Makefile
config.status: creating plugins/model_loader/Makefile
config.status: creating plugins/oscilloscope/Makefile
config.status: creating plugins/performance_measurement/Makefile
config.status: creating plugins/system_control/Makefile
config.status: creating plugins/userprefs/Makefile
config.status: creating deps/hdf/Makefile
config.status: creating include/rtxi_config.h
config.status: executing depfiles commands
config.status: executing libtool commands
developer@developer-HP-xw8600-Workstation:~/rtxi-master$ make
Making all in rtxi
make[1]: Entering directory `/home/developer/rtxi-master/rtxi'
  CXX    atomic_fifo.o
  CXX    cmdline.o
  CXX    daq.o
  CXX    default_gui_model.o
  CXX    event.o
  CXX    fifo.o
  CXX    io.o
  CXX    main.o
  CXX    main_window.o
  CXX    mutex.o
  CXX    plugin.o
  CXX    rt.o
  CXX    rtfile.o
  CXX    rwlock.o
  CXX    sem.o
  CXX    settings.o
  CXX    workspace.o
  CXX    rt_os-xenomai.o
../src/rt_os-xenomai.cpp: In function ‘long long int RT::OS::getTime()’:
../src/rt_os-xenomai.cpp:152:38: error: ‘rt_timer_tsc’ was not declared in this scope
  return rt_timer_tsc2ns(rt_timer_tsc());
                                      ^
../src/rt_os-xenomai.cpp:152:39: error: ‘rt_timer_tsc2ns’ was not declared in this scope
  return rt_timer_tsc2ns(rt_timer_tsc());
                                       ^
make[1]: *** [rt_os-xenomai.o] Error 1
make[1]: Leaving directory `/home/developer/rtxi-master/rtxi'
make: *** [all-recursive] Error 1

Any help would be greatly appreciated.

anselg, 2015-08-25:
Could you tell me what steps you've already taken with the installation? Particularly, did you compile and install Xenomai libraries? (You can check if the `/usr/xenomai` directory exists.) Also, which version of Xenomai are you using? 

----

One last question. Is there a reason why you are trying to compile RTXI with both the comedi and analogy drivers? Could you instead run: 
````
$ ./configure --enable-xenomai  --enable-analogy
````

topquant, 2015-08-26:
Thanks for getting back to me.  I was using the wrong version of xenomai and the wrong unix kernel.  Xenomai-3.0-rc6 does not support tsc only ns.  I created a new environment with linux 3.8.13 and xenomai 2.6.4. . So this problem is no loner an issue.

closed 2015-08-26 by yapatel