RTXI/rtxi #88: RTXI 2.0 freezes when configuring input channel

Wmdhampton, 2015-08-27:
I am following an old tutorial to try to get RTXI 2.0 up and running:

http://citeseerx.ist.psu.edu/viewdoc/download;jsessionid=85B5A1BF2B412AFABDF5CD972C4DE96C?doi=10.1.1.474.5372&rep=rep1&type=pdf

In section 3.2 it tells me to calibrate my DAQ card using:

$ comedi_calibrate

I disabled the comedi drivers, so this command does not work, and am using the analogy drivers as I was instructed to do in the custom installation option from the RTXI website:

http://rtxi.org/install/

This may be causing my next issue, so if I need to calibrate my DAQ card with the analogy drivers please tell me how.

When I open RTXI and select Control Panel from the system menu I select 
Channel: Input 0
All other settings are the same as the manual except that my Device is analogy0. However, when I click apply RTXI freezes. I have to click the close button, after which I am given an error message that RTXI is not responding and asking if I want to force quit. This issue does not occur when I apply setting for my Channel: Output 0

yapatel, 2015-08-27:
If you are attempting to manually install RTXI (ie without using the LiveCD), please refer to the documentation found [here](http://rtxi.org/install/) under "Custom".

The Analogy framework currently does not include calibration. We are working to add that in. The easiest way to solve that for now is to use a Windows computer to calibrate using NI-DAQmx.

When RTXI crashes, is there anything printed out at the terminal? If not, can you run rtxi using the following command:

sudo gdb rtxi (this will ask you for a password)

then enable Channel Input 0 and hit apply. If it crashes, an error should be printed to your terminal.

Wmdhampton, 2015-08-27:
When I type sudo gdb rtxi I get the message seen below. However, this does not open the regular rtxi window. Is there a way to enable channel Input 0 from the terminal while running gdb?

GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rtxi...(no debugging symbols found)...done.

yapatel, 2015-08-27:
It appears that GDB is working fine. You should see a prompt after the last line, at which you type "run" to start RTXI inside of the debugger.

Wmdhampton, 2015-08-27:
Here is the output:

Starting program: /usr/local/bin/rtxi 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[New Thread 0x7ffff7ea0700 (LWP 3539)]

Program received signal SIGXCPU, CPU time limit exceeded.
[Switching to Thread 0x7ffff7ea0700 (LWP 3539)]
0x00007fffed3e79a7 in rt_dev_ioctl () from /usr/xenomai/lib/librtdm.so.1

yapatel, 2015-08-27:
Do you have team viewer installed on your machine? If so, can you send the id/pass to yogi at rtxi.org?

It appears there is an issue with the user/group settings.

Wmdhampton, 2015-08-28:
I don't have team viewer installed. If you tell me how to install it I would be happy to do so.

anselg, 2015-08-28:
In your RTXI repository, there is a script to install TeamViewer. It's called install_teamviewer.sh and is in the scripts/dev folder. Just run:
````
$ ./install_teamviewer.sh
````

Just hit "y" whenever appropriate, and TeamViewer will be installed. Open in up from the applications menu. Two numbers will show up. One is an ID for your installation, and another is a password for the session. Email them to Yogi (DON'T POST THEM HERE), and as long as TeamViewer is open and running, he'll be able to log into your system.

yapatel, 2015-09-01:
This is now resolved.

The problem is hardware-based (DAQ appears to be malfunctioning).

closed 2015-09-01 by yapatel