RTXI/rtxi #89: Can't change default boot kernel for Ubuntu 14.04 to Xenomai

Wmdhampton, 2015-08-27:
After installing the custom build of RTXI 2.0 I am following the instructions for changing the default boot kernel on the RTXI site:

http://rtxi.org/docs/tutorials/2015/04/01/setting-a-default-boot-kernel/

Unfortunately neither of the two suggested options work. I am still having to press shift during boot-up, and manually selecting my Xenomai kernel, otherwise the wrong kernel boots. If anyone has had a similar experience please let me know.

yapatel, 2015-08-27:
Please run "sudo update-grub" and paste the output here.

Wmdhampton, 2015-08-27:
Warning: Setting GRUB_TIMEOUT to a non-zero value when GRUB_HIDDEN_TIMEOUT is set is no longer supported.
Found linux image: /boot/vmlinuz-3.16.0-30-generic
Found initrd image: /boot/initrd.img-3.16.0-30-generic
Found linux image: /boot/vmlinuz-3.8.13-xenomai-2.6.4
Found initrd image: /boot/initrd.img-3.8.13-xenomai-2.6.4
Found memtest86+ image: /boot/memtest86+.elf
Found memtest86+ image: /boot/memtest86+.bin
done

yapatel, 2015-08-27:
Great; these are zero-indexed, so to get your xenomai kernel, you should set the following line:

GRUB_DEFAULT="1>2"

Then run "sudo update-grub", and that should reboot you into the xenomai kernel automagically.

Wmdhampton, 2015-08-28:
Unfortunately this is not working. Please let me know if you have any either ideas. Luckily this is not critical for operations. Just a little annoying.

anselg, 2015-08-28:
Could you paste the lines that aren't commented out from your /etc/default/grub file here? 

There are basically two ways to set a default kernel:  

1. Change `GRUB_DEFAULT=0` to `GRUB_DEFAULT=saved` and then add a new parameter `GRUB_SAVEDEFAULT=true`. After you run update-grub, reboot and manually select the kernel from the menu. From now on, whenever you boot, it will default to the kernel last used, so as long as the last kernel was the Xenomai one, you won't need to go through the boot menu. 
2. Change `GRUB_DEFAULT=0` to `GRUB_DEFAULT="1>2"`. If you do this, do not define the `GRUB_SAVEDEFAULT` parameter. Run update-grub, and the default kernel will now permanently set to the Xenomai kernel (the 3rd option that comes up in the "Advanced options..." submenu) .

I'll add my /etc/default/grub file here for reference. 

````
GRUB_DEFAULT=saved
GRUB_SAVEDEFAULT=true
GRUB_HIDDEN_TIMEOUT=0
GRUB_HIDDEN_TIMEOUT_QUIET=true
GRUB_TIMEOUT=10
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash crashkernel=384M-:128M"
GRUB_CMDLINE_LINUX=""
````

Wmdhampton, 2015-08-28:
Ah, I finally realized my issue. I thought that setting GRUB_DEFAULT=saved would automatically load the last kernel that was installed. I tried this again an ran the correct kernel, and then my computer loaded the correct kernel on the next boot. Thanks!

closed 2015-08-28 by yapatel