RTXI/rtxi #90: Axoclamp-2A amplifyer giving strange noise

Wmdhampton, 2015-08-27:
When ever I connect my:
 
DAQ Analog Output 0 --> Axoclamp EXT. ME1 COMMAND

My I(nA) display reads that I am injecting 0.6(nA) even though I am not. This goes away as soon I unplug the DAQ Analog Output 0 on my BNC-2090. I though this might be a strange ground loop error. However, this does not occur when the Axoclamp EXT. ME1 COMMAND is connected to a different computer through a Digidata 1440A. Both of my computers are grounded through the same circuit breaker. Any thoughts would be greatly appreciated.

yapatel, 2015-08-27:
A few questions...

1. Which breakout board are you using with your setup? If you are using a BNC-2110, first make sure that the switches under each AO post are set to FS. 

2. Does this happen when you disconnect the DAQ Analogy Output --> Axoclamp connection?

3. Does this happen when you connect to other AO channels (ie AO 1)?

Wmdhampton, 2015-08-28:
The DAQ card that I am using is the NI PCI-6052E:
http://sine.ni.com/nips/cds/view/p/lang/en/nid/2601
on the BNC-2090 there are switches that can be set to SE or DIFF. The problem occurs in both settings. When I disconnect the DAQ Analogy Output from the Axoclamp the problem goes away. This occurs on channel 0 and 1 for the output from the BNC-2090. However, it does not occur for any of the input channels.

yapatel, 2015-09-01:
This is now resolved.

The problem is hardware-based (DAQ appears to be malfunctioning).

closed 2015-09-01 by yapatel