RTXI/rtxi #92: Linking c files

tnetoff, 2015-09-02:
I have two files that I want to compile and use to make a module.
I have included the headers and the c files in the Makefile.  The files seem to compile.  But, when I run the module it says it cannot find the functions that are defined in the second c file.  I feel like I am not linking properly.
Here is the Makefile:

PLUGIN_NAME = Arduino

HEADERS =  arduino_serial.h Arduino.h 

SOURCES = arduino_serial.c Arduino.cpp 

LIBS = 

### Do not edit below this line ###

include $(shell rtxi_plugin_config --pkgdata-dir)/Makefile.plugin_compile

yapatel, 2015-09-02:
Can you please copy paste the error you get here?

tnetoff, 2015-09-02:
Hey Yogi,
So, I have a c file and a cpp file for making the module.
It does not seem to be compiling the c file.  I see .la .lo .o files for
the .cpp file but nothing for the .c file.
I have added some lines in to deal with the c file
such as:
#ifdef _cplusplus
extern "C" {
#endif

in the c file header and

extern "C"{
#include "arduino_serial.h"
}
in the cpp header

Below is the error.  The function I want to use, which is defined in the c
file, is serialport_read_until:

comedi_device.cpp:42:ComediDevice::ComediDevice : Failed to parse comedi
soft calibration file/n../src/plugin.cpp:51:Plugin::load : failed to load
Arduino.so: /usr/local/lib/rtxi/Arduino.so: undefined symbol:
serialport_read_until

If you want to log in and play, this is the ip address: 128.101.203.249

Unfortunately, this is the old version of RTXI since the new version has
been moved over to our experimental rig and we have not updated this dev
machine.  On my todo list.

Thanks,
Tay

tnetoff, 2015-09-02:
Hey Yogi,
Here are all the files.
Tay

tnetoff, 2015-09-02:
Hey Yogi,
Here are all the files.
It is a module to read the com port to communicate with an Arduino.
Tay

closed 2015-09-09 by yapatel