RTXI/rtxi #96: Istep module not delivering current pulses

Wmdhampton, 2015-10-23:
I am trying to work through the tutorial in section 3.2 of the RTXI manual: http://citeseerx.ist.psu.edu/viewdoc/download;jsessionid=85B5A1BF2B412AFABDF5CD972C4DE96C?doi=10.1.1.474.5372&rep=rep1&type=pdf. I have loaded and configured the Istep module as specified. I have also followed the steps involved with the System Control Panel and the Oscilloscope. All of the cables are plugged into the proper place on my axoclamp-2A amplifyer, which is connected to a model cell. There does not appear to be any problems with the amplifyer, as I am able to apply current without any problems. When I try to unpause the Istep module nothing happens. I'm not sure what the problem is. Any ideas?

yapatel, 2015-10-24:
Hi Dave -

Could you please tell me which version of RTXI you are using (Help Menu->About RTXI)

Also, can you please elaborate on what you mean by "nothing happens"?

Yogi

Wmdhampton, 2015-10-26:
I am using version 2.0. When I said that nothing happens I meant that I am not getting a reading on my oscilloscope, and that there are no current injections being sent through the amplifier. Essentially, I am expecting the protocol to run, but this not not seem to be occurring.

yapatel, 2015-10-26:
Do you see the output of the current-step module on the RTXI 
oscilloscope? Make sure you set the scale of the Iout channel in the 
oscillscope to something like 2pV/div.

Wmdhampton, 2015-10-26:
I don't see anything on the RTXI oscilloscope, even with the scale that you recommended.

yapatel, 2015-10-26:
Can you install teamviewer and email your ID and pass to yogi <at> rtxi.org?

See https://github.com/RTXI/rtxi/issues/88#issuecomment-135798709 if you 
need help installing teamviewer.

Please do not send those credentials to this github post.

yapatel, 2015-10-26:
Resolved.

For other users: please note that you need to set your ground settings in RTXI appropriately. If you are using a BNC-2110 breakout board, your measurements are all "Differential". If you are using a BNC 2090, your measurements could be either "Differential", "Common", "Ground" - depending on what you have your breakoutboard set to. If "RSE", please set to Ground. If NRSE, please set to "Common".

closed 2015-10-26 by yapatel