RTXI/rtxi #97: invalid pointer and corrputed double-lined list errors

neuroking, 2015-10-29:
Hello,

One of my module crashes unexpectedly while running. I seem to get these errors consistently:

*** Error in `rtxi': free(): invalid pointer: 0x00000000010a4bc0 ***
*** Error in `rtxi': corrupted double-linked list: 0x00000000011ffd50 ***

I was hoping I could get some insight as to what these errors are referring to.

Thanks

Vivek

yapatel, 2015-10-29:
Hi Vivek -

Can you run rtxi in gdb by running “sudo gdb rtxi”, then type “run” at the gdb prompt. Run your setup as you normally and then when it crashes, copy paste the output of gdb here?

Yogi

neuroking, 2015-10-29:
First time I tried to run the saved workspace I got this:
*** Error in `/usr/local/bin/rtxi': free(): invalid pointer: 0x0000000000a0dd30 ***

Program received signal SIGABRT, Aborted.
0x00007ffff5923cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.

On the second try I got the module to work. I had no problems what so ever. Then when I closed out of rtxi I got this message:
*** Error in `/usr/local/bin/rtxi': free(): invalid pointer: 0x00000000006fb0a0 ***

Program received signal SIGABRT, Aborted.
0x00007ffff5923cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.

Vivek

yapatel, 2015-10-29:
It sounds like your module isn’t properly handling “new” variables.

Please package your code with the makefile and send it to yogi at rtxi dot org.

Yogi

yapatel, 2015-10-29:
Took a quick look at your code - there's a big issue. You're calling "new" numerous times and declaring fairly large multidimensional arrays without ever calling "delete". Note that when you dynamically allocate space with new/malloc/etc, you need to call the respective delete/free/etc functions to prevent memory leaks and dangling pointers, which is why you are getting that error.

Your RefLearn destructor should look like:

```
159 RefLearnV::~RefLearnV(void) {
160   delete[] PCcoeffs1;
161   delete[] PCcoeffs2;
162   delete[] Freqs;
163   delete[] AlphaVec;
164   delete[] FreqActions;
165   delete[] ActionCosts;
166   delete[] Q;
167   delete[] et3;
168   delete[] ActionBank;
169   delete[] SMax;
170   delete[] Q_st;
171   delete[] cumprob;
172   delete[] SignalHist;
173   delete[] RewHist;
174 }
```

Then, you need to make sure that your arrays are large enough for what you are doing. Again, at a quick glance, it looks like you may occasionally be trying to access memory locations that do not exist/are not allocated.

neuroking, 2015-10-29:
Hey Yogi,
Thanks for the reply. I had no idea we needed to delete the memory allocation within the destructor. This is the first I've seen it haha. I'm a terrible programmer...but I'm learning...slowly.

I added the lines into the destructor. At first it seemed to work. Now when I try to load the settings file rtxi immediately crashes and I get this error:
*** Error in `rtxi': free(): invalid pointer: 0x0000000000ca0870 ***

Anyways, I then tried to run it in gdb mode. I got this error when I started pressing buttons on the RefLearnV module:
*** Error in `/usr/local/bin/rtxi': corrupted double-linked list: 0x00000000006fb0c0 ***

Program received signal SIGABRT, Aborted.
0x00007ffff5923cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56    ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
I've attached the code again. I'm not sure if I implemented your suggestion correctly.
Thanks Yogi! Vivek NagarajNeuroscience PhD CandidateGraduate Program in NeuroscienceUniversity of Minnesota - Twin Cities214-693-1585

yapatel, 2015-10-30:
Sending files to github posts does not work. Please email files separately to yogi at rtxi dot org.

Yogi

closed 2015-11-12 by yapatel