RTXI/rtxi #101: Error when loading module using linear algebra Eigen library

neuroking, 2016-02-26:
Hello,

I am getting the following error message when I try to load my module into rtxi:
../src/plugin.cpp:51:Plugin::load : failed to load LKF.so: /usr/local/lib/rtxi/LKF.so: undefined symbol: _ZTV3LKF

I do not get any errors compiling the module. 

I am not sure what to do at this point. 

Vivek

yapatel, 2016-02-26:
This means that you are not properly linking the Eigen library to your module; 

1. make sure your library is properly installed
2. the library's location is either in your search path
3. you are including the appropriate linker flag (i.e. -lEigen) in your plugin's makefile

neuroking, 2016-02-26:
Hi Yogi,

It is not necessary to install Eigen to use the library functions. That being said is it necessary to install it for use with Rtxi?

All the supporting library functions I need to run my module are in subfolder within the module folder. 

This is the Makefile configuration:

PLUGIN_NAME = LKF

HEADERS = LKF.h

SOURCES = LKF.cpp

INC = -I/RTXI/NetoffModels/LKF/Eigen

LIB = -L/RTXI/NetoffModels/LKF/Eigen

### Do not edit below this line ###

include $(shell rtxi_plugin_config --pkgdata-dir)/Makefile.plugin_compile

anselg, 2016-03-10:
This thread seems dead, and #102 implies you got around the issue. I'll go ahead and close this. Please reopen if the issue persists.

closed 2016-03-10 by anselg