RTXI/rtxi #103: Support for NI X series cards?

adamltaylor, 2016-04-26:
What's the status of support for NI X series cards these days?  I know it was a work-in-progress at some point...

anselg, 2016-04-27:
The short of it is that it's still a work in progress. 

We use a third-party driver, called analogy, developed by Xenomai, the same people that develop the real-time kernel. Right now, they haven't built a driver that supports X series cards, but we've sent them a card that they're using to develop one. 

The main hurdle is that for X series cards, NI altered their license agreement so that it's difficult to develop and distribute open source drivers. The terms allow for GPL'ed code under stringent conditions, but it's nebulous about when those conditions are met. See: [https://decibel.ni.com/content/docs/DOC-16444](https://decibel.ni.com/content/docs/DOC-16444) and [this thread](https://groups.google.com/forum/#!msg/comedi_list/g2SDCV7bcQs/l0rvXEFBPjAJ). 

It's possible to use [NI's driver development kit](http://sine.ni.com/nips/cds/view/p/lang/en/nid/11737) to build a driver on your own. We aren't working on anything like that with X series cards. Right now, most of our development is focused on expanding RTXI for new research applications (high channel counts, RT imaging, etc.). We're NIH-funded, after all. For now, the plan is to continue relying on Xenomai developers to add support for X series cards. 

I should also add that down the road, we'll be moving away from PCI devices for data acquisition and will instead rely on an open-source (hardware and software) embedded computer that will function as both the DAQ and the RTOS. We have a proof of concept project, called [Puggle](http://www.open-ephys.org/puggle/), based on a BeagleBone Black.

yapatel, 2016-04-27:
Some slight additions/modifications...

Our initial efforts with developed an X series driver (ourselves) were halted due to a lack of register maps and necessary information. When this information was released in the NIDDK, the integrations of X-series registers and drivers into the Xenomai+Analogy framework was halted because we discovered that the X-series card did not have the NI-MITE ASIC anymore - and instead has a new proprietary ASIC that does not have complete open-source documentation available (AFAIK, last checked in January 2016). Note that the examples and code provided in the NIDDK for the X-series are not directly integratable into RTXI, because the driver interrupts have to go through the Xenomai/Analogy framework for RT control.

Xenomai has stated that although developing an X-series driver is of interest, it is a significant overhaul and thus, it is not at the top of the "to-do" list. This led us down the path of developing an alternative system that is based upon single board computers.

closed 2016-07-08 by yapatel