RTXI/rtxi #104: Error when launching rtxi from terminal

neuroking, 2016-06-08:
Hello,

We get the following error message when we type in rtxi or sudo rtxi into the command terminal:
Xenomai: /dev/rtheap is missing
(chardev, major=10 minor=254)

We have the latest version of rtxi installed. 

What do I need to do to fix this? 

Thank you

Vivek

yapatel, 2016-06-08:
You're not booted into the RT kernel. You can check which kernel you are booted into by typing "uname -r" at the command line. 

When your computer boots, hold the right shift key to get the grub menu and select the kerne with "Xenomai" in the name. 

Yogi

anselg, 2016-06-08:
What is the output of 'uname -r' ?

You may not be running a real-time kernel. See FAQ #1:
http://rtxi.org/docs/#faqs

anselg, 2016-06-14:
It was a simple question, so I'm guessing it's resolved by now. For future reference, please close thread if your issue is fixed.

closed 2016-06-14 by anselg

neuroking, 2016-06-14:
Gotcha, thanks for the fast answer! I'll make sure to close any future threads after they are completed .

Vivek

oPiZiL, 2019-09-12:
I met the same problem.   
`uname -a`  
``` 
Linux louw-desktop 4.9.51-xenomai-3.0.5-aufs #1 SMP Sat Dec 23 20:47:47 EST 2017 x86_64 x86_64 x86_64 GNU/Linux
```