RTXI/rtxi #106: Unable to extract data from .h5 file

neuroking, 2016-06-29:
Hello,

I am unable to open a .h5 file. RTXI crashed while I was recording data. The file is 90.33MB so I know there is data in the file, but I am unable to extract it using the MATLAB hdf5 extraction files. 

What should I do?

Thank you

Vivek

anselg, 2016-06-29:
Could you tell me: 

1. What are the permissions for your data file? Run `ls -al` in the directory where the file is and paste the output line with your data file here. 

2. What sort of error message are you getting when you run the MATLAB script? What function are you using?

neuroking, 2016-06-29:
Hi Ansel,

1. The permissions for the file I am trying to open are the same as all the other files
-rw-r--r--

2.  Error message:
Error using hdf5infoc
Call to HDF5 library failed (invalidFile): "Not a valid HDF5 file:
U:\InVivoExperiments\LQG\06192016\06192016_LQGC_4AP_nostim_01.h5".

Error in hdf5info (line 78)
[info.Offset, info.GroupHierarchy, majnum, minnum, relnum] = ...

Error in rtxi_read (line 9)
fileinfo = hdf5info(fname); % load default information from built-in MATLAB function

anselg, 2016-06-29:
Hm. There should be an application called hdfview installed on your system. (If not, run `sudo apt-get install hdfview`). Try to open that file in hdfview (run `hdfview yourfile.h5` in the terminal). 

A UI will pop up that will allow you to explore the data via the HDF metadata. Try to see 1. if hdfview can open the file and 2. if the data for the last trial is in some way corrupted. If yes to both, you can use the right-click menu to delete trials. Save the changes (in a different file) and then try to open it in MATLAB.

yapatel, 2016-07-08:
Was this resolved?

neuroking, 2016-07-09:
Hi Yogi,
Sorry for not replying. I have not had the opportunity to address this issue. Been preoccupied with a multitude of other projects. I will try your recommendations first thing next week. If I can't figure it out I'll send you the .h5 file.
Vivek
 Vivek NagarajNeuroscience PhD CandidateGraduate Program in NeuroscienceUniversity of Minnesota - Twin Cities214-693-1585

yapatel, 2016-08-18:
Assuming this was resolved...

closed 2016-08-18 by yapatel