RTXI/rtxi #107: RTXI and Neuron Model Troubleshooting

rythorpe, 2016-07-07:
I have been trying to update our data retrieval/analysis system to RTXI and started by trying to run the neuron module. I am now concerned that the issue may be with my installation of RTXI rather than the neuron module itself. We have struggled to get the neuron module to give even a baseline that was visible on the oscilloscope. After a bit of troubleshooting I decided to run RTXI off of the Ubuntu live CD because I had a theory that some of the Qt includes were not accessible do to damaged directory pathways (whether this is true or not, I'm still not certain). After pulling up RTXI and installing the appropriate modules, I was able to get neuron to run as expected, evident by a nice action potential displayed on the oscilloscope. I then tried running the install_dependencies.sh in the script file. Restarting RTXI and reestablishing my previous module settings, Vm increased exponentially on the oscilloscope. I immediately increased my field of view as much as possible to 10 V/div and 5 s/div and immediately found the signal again. Vm was still increasing dramatically until it again left my screen.

anselg, 2016-07-07:
I have not been able to reproduce the issue with the neuron module. 

From your post, I tried the neuron module: 

1. From the Live CD (works)
2. Immediately after installing RTXI from the Live CD (works)
3. After running ./install_dependencies.sh after installing from Live CD (works)
4. After running ./install_rtxi.sh after running ./install_dependencies (works)

By "works," I mean that I can inject current (a periodic square wave) into the model and trigger a stable train of action potentials with believable morphology. 

Could you try these things (from your RTXI installation, not the Live CD): 

##### 1. Save the workspace you are using to test the neuron module and post it here. 

Open the modules you were running and hit "Save workspace" the Files menu. (http://rtxi.org/docs/manual/#menus). A workspace file contains an XML-formatted list of your RTXI settings. 

##### 2. Check whether the neuron or oscilloscope is producing the anomalous behavior. 

Output the neuron module to a hardware oscilloscope and see if there's still a steady increase in output. 

If that's not possible, another option is to record the output of the neuron module in the data recorder and see cell potential is steadily increasing. 

To visualize the output, you can load the recorded data in the analysis tools module. Install it by running: 
```
git clone https://github.com/rtxi/analysis-tools
cd analysis-tools
git checkout analysis-module
make -j`nproc` && sudo make install
```
Analysis-tools is a work-in-progress module that plots recorded data. Open the module, load the HDF file, and hit the "Plot" button. If the plots show that steady increase in potential, then it's an issue with the neuron module, not the oscilloscope. Let me know if you need more detailed instructions. 

You can also check other cell models. Try the Connor-Stevens or Wang-Buzsaki models and see if you see the same error in the oscilloscope.  

##### 3. Reinstall RTXI. 

In case there actually is some damaged dependency path, reinstall RTXI. From the scripts folder of the RTXI repo, run: 
```
./uninstall_rtxi.sh
./install_dependencies.sh
./install_rtxi.sh  # Pick option 1 - Xenomai + analogy
```

yapatel, 2016-07-08:
There were dependency issues on your system; a clean reinstall of dependencies+rtxi did the trick.

closed 2016-07-08 by yapatel

rythorpe, 2016-07-12:
As of Monday, our system magically stopped working and reverted back to what it was doing before. By this point, I am assuming the problem is on my end, however, I have tried uninstalling/reinstalling RTXI, reformatting my hard drive and reinstalling from the live CD yet the problem persists. I have know idea what is going on or what to do next. Any ideas? Could I possibly get some info on what hardware you guys are using?

anselg, 2016-07-12:
The stats page (http://rtxi.org/stats/) has specs for two machines running RTXI. 

Do you see that behavior in modules other than the hh neuron, like connor-stevens, faber-rudy, etc.?