RTXI/rtxi #108: Hdf5 to python

rythorpe, 2016-07-08:
We'd like to use python for our data analysis, but can't read the HDF5 version supported by RTXI via pytables and pandas. Could RTXI be updated to output the most recent version of HDF5?

Here is our error report from python:

ValueError: PyTables [3.2.3.1] no longer supports opening multiple files
even in read-only mode on this HDF5 version [1.8.4]. You can accept this
and not open the same file multiple times at once,
upgrade the HDF5 version, or downgrade to PyTables 3.0.0 which allows
files to be opened multiple times at once

anselg, 2016-07-08:
Try the h5py library. (sudo apt-get install python-h5py)

anselg, 2016-07-08:
Here's a quick snippet that will read an trial from an recorded hdf file as a pandas dataframe and plot the timerseries. (copied from the getTrial.py function in my handy-scripts repository; not tested much): 

```
#! /bin/python

# Requires h5py, seaborn

import h5py as h5
import numpy as np
import pandas as pd
import seaborn as sb
from matplotlib.backends.backend_pdf import PdfPages

filename = "test.h5"
plotname = "test.pdf"
trialnum = 1

def getNumTrials(f):
    ntrials = len(f["/"])
    print("# of Trials:\t", ntrials)
    return ntrials

def getTrial(f, n):
    trialname = "/Trial" + str(n)
    return f[trialname]

def getPeriod(f, n):
    return f["/Trial" + str(n) + "/Period (ns)"].value

def getDownsamplingRate(f, n):
    return f["/Trial" + str(n) + "/Downsampling Rate"].value

def getTrialLength(f, n):
    return f["/Trial" + str(n) + "/Trial Length (ns)"].value

def getChannelFrame(f, n):
    meta = [ item for item in f["/Trial" + str(n) + "/Synchronous Data"] ]
    headers = [ item.split(" : ")[1] for item in meta[0:3] ]
    data = f["/Trial" + str(n) + "/Synchronous Data/" + str(meta[len(meta)-1])]
    frame = pd.DataFrame(data=np.vstack(data), columns=headers)
    frame["Time (ms)"] = range(0, len(frame))
    return frame

hdf = h5.File(filename, 'r')

getNumTrials(hdf)

period = getPeriod(hdf, trialnum) # in ns
downsampling = getDownsamplingRate(hdf, trialnum)
length = getTrialLength(hdf, trialnum) # in ns
frame = getChannelFrame(hdf, trialnum)
meltyframe = pd.melt(frame, "Time (ms)")

hdf.close()

# gennerate plots
with PdfPages(plotname) as pdf:
#    p = sb.lmplot(x="Time (ms)", y="value", hue="variable", data=meltyframe, size=5, aspect=2, fit_reg=False)
#    pdf.savefig(p.fig)

    for field in meltyframe['variable'].unique():
        p = sb.lmplot(x="Time (ms)", y=field, aspect=2, size=5, data=frame, fit_reg=False)
        pdf.savefig(p.fig)

```

closed 2016-09-06 by anselg