RTXI/rtxi #109: Oscilloscope trigger/crosshair

atzbacher, 2016-08-18:
Dear all,

I am totally new to the world of rtxi and have a few problems/question. 
Installing the OS form the live image and so on was a piece of cake. Data acquisition was working out of the box and the application itself is really user friendly compared to other commercially available solutions in my field.

However I experienced a few (minor) issues with the oscilloscope. I had a look at the source code, and it seems to me that a crosshair has been implemented already(My C++ knowledge is still almost not existent, so I might be wrong here). However, I can't see any crosshair at the oscilloscope while running the application.
Is the code included in the live image a older version which I have to update or did I made a mistake during or after setting up the system?

The second issue with the oscilloscope is the triggering. The triggering seems to recognise the threshold. However it only displays the first few ms of after the trigger and does not allow me to change the scaling whatsoever. 

The last issue I have concerns the axon Multipatch 700 controller. The output scaling values are not correct. Using the controller to set the scaling and the membrane test plugin  to calculate the resistance over a 10 M0hm resistor gives me 25 MOhm calculated by the plugin.
Attaching an external oscilloscope gives me a voltage step of 200mV for a 10 mV step in the membrane test plug in. However, the output should be 500 mV for a 10 mV and an input sensitive of 20mV/V for the 700 B (after that adjustment the resistance is calculated correctly, as is the 33 pF capacitor and the 500 MOhm and 10 MOhm membrane and series resistance of the model cell in cell config)
The same is true for the IClamp scaling.  (2.5 V/nA rather than 400pA/V)
Is that a real issue, or did I miss something?
The input scaling seems to be correct after testing the response to an external waveform. 

Thanks for maintaining this great project,  and sorry for the n00b questions 

best
pS

yapatel, 2016-08-18:
Both problems were fixed in the [qt5](https://github.com/rtxi/rtxi/tree/qt5) branch. Please update your installation by doing the following:

```
cd /home/RTXI/rtxi
git checkout qt5
git pull
cd /home/RTXI/rtxi/scripts/
sudo ./uninstall_rtxi.sh
sudo ./install_dependencies.sh
sudo ./install_rtxi.sh
->select the Xenomai+Analogy configuration when asked
```

Let us know if you have any questions.

atzbacher, 2016-09-06:
Hi,

I finally found some time to update to the qt5 branch. Which indeed resolved the issues with the oscilloscope. 

The problem with the 700B scaling plug-in persists.

best
pS

anselg, 2016-09-06:
You're right. That's a bug with the amp controller. The issue was it was using the values for the sensitivities (e.g. 20, 100) as the actual gains without properly scaling them (e.g, 1/20e-3, 1/100e-3). I pushed a fix. Just `git pull` your clone of the axon-multiclamp repository to download it and then recompile. 

As the other two issues are resolved now, I'll go ahead and close this thread. I've opened up another on the multiclamp700 repo (https://github.com/RTXI/axon-multiclamp700/issues/1) should you still have issues.

closed 2016-09-06 by anselg