RTXI/rtxi #110: Trouble installing RTXI on Core i5 PC

yamamoto-h, 2016-08-24:
Hi,
I am having trouble installing RTXI on a new computer (Intel Core i5-6600K) and was wondering if I could get any help.

I first tried installing from the live RTXI 2.0 image (amd64) in USB, but after selecting "Install Ubuntu GNOME" in GRUB menu, an error message "Fast TSC calibration failed" then "Kernel panic - not syncing" appears.

I followed the instructions for the custom install-- but in this case, I cannot launch Linux 3.8.13-xenomai-2.6.4. I get an error message "Loading initial ramdisk" and then the system freezes.
The same message appears in the recovery mode as well. 

I found a post reporting a similar issue in xenomai.org (for Linux 3.14.17), but wasn't sure if it was related--
http://www.xenomai.org/pipermail/xenomai/2015-May/034274.html

Thank you in advance.

Hideaki

anselg, 2016-08-24:
Could you paste the output of "cat /proc/cpuinfo" here? ( Wrap the entire
block in ```.) That will give a bit more information on your CPU.

yamamoto-h, 2016-08-24:
Thank you for the prompt response. Here is the cpuinfo--

`````````````````````````````````````````````````````````````````````````````````
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 94
model name	: Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
stepping	: 3
microcode	: 0x84
cpu MHz		: 3810.214
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch ida arat pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bugs		:
bogomips	: 7004.87
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 94
model name	: Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
stepping	: 3
microcode	: 0x84
cpu MHz		: 3831.816
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch ida arat pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bugs		:
bogomips	: 7004.87
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 94
model name	: Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
stepping	: 3
microcode	: 0x84
cpu MHz		: 3896.210
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 2
cpu cores	: 4
apicid		: 4
initial apicid	: 4
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch ida arat pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bugs		:
bogomips	: 7004.87
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 6
model		: 94
model name	: Intel(R) Core(TM) i5-6600K CPU @ 3.50GHz
stepping	: 3
microcode	: 0x84
cpu MHz		: 1915.703
cache size	: 6144 KB
physical id	: 0
siblings	: 4
core id		: 3
cpu cores	: 4
apicid		: 6
initial apicid	: 6
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch ida arat pln pts dtherm hwp hwp_noitfy hwp_act_window hwp_epp intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1
bugs		:
bogomips	: 7004.87
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
``````````````````````````````````````````````````````````````````````````````````````````

anselg, 2016-08-24:
The likely issue is that the CPU is new (~late 2015) and the kernel is old
(~August 2013). Drivers and firmware needed for newer hardware isn't
present in older kernels.

The next thing to try is to install an rt kernel based on a newer standard
one. Will update with instructions soon.

yamamoto-h, 2016-08-24:
Thanks. I tried modifying the install_rt_kernel.sh to have it load a newer kernel, but couldn't get it to run properly.

I look forward to the updated instructions.

anselg, 2016-08-24:
I uploaded a script that will install a newer kernel. Download it and run it as you did the `install_rt_kernel.sh` script earlier. (Make sure the script is executable before running `./install_rt_kernel.sh` or just run `bash install_rt_kernel.sh`.) 

https://gist.github.com/anselg/9cbc6db00bd4626b712db5ed059e3424

The script will install an rt kernel based on 3.18.20. If the problem persists, we can move to kernel 4.1.18. That will also involve some additional steps, but I'll describe them should it come to that.

yamamoto-h, 2016-08-25:
Thanks for the new script. I clean-installed Ubuntu 14.04.4 amd64 on the PC and then went through the custom install again.
With the new script, I was able to install 3.18.20-xenomai-2.6.5, which I could boot without the previous error (i.e., freezing with "Fast TSC calibration failed").
But now I have new issues--

1. When I continue the RTXI installation with ```sudo ./install_rtxi.sh```, the process terminates with a compilation error on rt_os-xenomai.cpp saying that there is no native/task.h.
```
Making all in rtxi
  CXX      rt_os-xenomai.o
../src/rt_os-xenomai.cpp:25:25: fatal error: native/task.h: No such file or directory
 #include <native/task.h>
                         ^
compilation terminated.
make[1]: *** [rt_os-xenomai.o] Error 1
make: *** [all-recursive] Error 1
----->RTXI compilation failed.
```

2. Some error messages are displayed at start-up of the OS-- "xsave: eagerfpu not present, disabling some xstate features: 0x18", "Ignoring BGRT: invalid status 0 (expected 1)", and "kvm: disabled by bios". The messages are stored in /var/log/kern.log

The resolution of the LCD is also not right (and cannot be modified in System Settings) when I boot the system with 3.18.20-xenomai-2.6.5.

I appreciate any advice. Thank you.

yapatel, 2016-08-25:
It looks like the Xenomai userspace libraries are not installed. The following will re-install the libraries.

```
cd /opt/build
sudo ./configure --enable-shared --enable-smp --enable-x86-sep
sudo make install
'''

Then rerun ```sudo ./install_rtxi.sh"

yamamoto-h, 2016-08-25:
/opt/build directory was empty. So I tried the configure/make in /opt/xenomai-2.6.5.
After that rtxi intallation completed, and an executable files were generated under /usr/local/bin/.
The problem now is that no window pops up after running ```sudo rtxi```. In the ```top```, rtxi appears for ~1 s, then disappears.

Here is the output of ```lspci -vvv | grep VGA```--
```
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
00:02.0 VGA compatible controller: Intel Corporation Sky Lake Integrated Graphics (rev 06) (prog-if 00 [VGA controller])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
```

Another thing that I am seeing now is that, after running configure and make in /opt/xenomai-2.6.5, a window saying "System program problem detected" is popping up intermittently. 
ProblemType: Kernel Oops
OopsText: BUG: unable to handle kernel paging request at ...

Thank you.

yapatel, 2016-08-25:
Is there anything printed to the terminal when you run ```sudo rtxi``` and it closes?

Can you also provide the output of ```dpkg --list | grep linux-image```?

Regarding the graphics, do you have a dedicated graphics card you can use (PCI based, recommended) rather than the integrated graphics on the computer?

yamamoto-h, 2016-08-25:
After running ```sudo rtxi``` (and typing in the password), the terminal just hangs up and nothing is printed to the terminal.

Here is the output of ```dpkg --list | grep linux-image```--
```
ii  linux-image-3.18.20-xenomai-2.6.5         20160825                                amd64        Linux kernel binary image for version 3.18.20-xenomai-2.6.5
ii  linux-image-4.2.0-27-generic              4.2.0-27.32~14.04.1                     amd64        Linux kernel image for version 4.2.0 on 64 bit x86 SMP
ii  linux-image-4.2.0-42-generic              4.2.0-42.49~14.04.1                     amd64        Linux kernel image for version 4.2.0 on 64 bit x86 SMP
ii  linux-image-extra-4.2.0-27-generic        4.2.0-27.32~14.04.1                     amd64        Linux kernel extra modules for version 4.2.0 on 64 bit x86 SMP
ii  linux-image-extra-4.2.0-42-generic        4.2.0-42.49~14.04.1                     amd64        Linux kernel extra modules for version 4.2.0 on 64 bit x86 SMP
ii  linux-image-generic-lts-wily              4.2.0.42.34                             amd64        Generic Linux kernel image
```

At the moment I don't have a dedicated graphics card with me, but I can surely get one if necessary.

Thanks.

yapatel, 2016-08-25:
Can you email me at yogi at rtxi dot org? We're going have to do a bit of customization to get things working.

I'll post what we do here after we're done for others facings similar issues (until the newer RT kernels are stable).

wosen7, 2017-02-01:
FYI, I had a problem with installing RTXI on a new Acer Aspire R 11. When running the install_rt_kernel.sh I got the error message that several modules' config files could not be found and it would crash before reaching the kernel configuration menu. However, when using a newer kernel by using the link posted above by yapatel it worked. So, thank you very much!

(Just wanted to drop this note in case someone else runs into the same problem)

yapatel, 2017-02-02:
Glad you got it and thanks for posting the feedback!

closed 2017-06-24 by yapatel