RTXI/rtxi #112: Low performance during Xenomai latency test

cpalvarezb, 2017-06-12:
Hi, we are setting up a new dynamic clamp system in our lab. I have tried two computers (old processors) to see if the performance was good enough, but the latency tests show long latencies compared to the example you have in the installation page. I have uploaded the two performance plots. Would you recommend moving to a more recent processor and better graphics card?

[image]
[image]

Thank you,

Claudia

 

## Basic Information: 
**RTXI Version:** 
**Installed from Live CD?:**

## Error Description: 

**What happened?**

**What was supposed to happen?**

**What steps will trigger the error?**

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```

```

yapatel, 2017-06-13:
How are you installing RTXI? Is this with the LiveCD or a manual installation?

cpalvarezb, 2017-06-13:
Hi, thank you for your reply. I installed RTXI following the directions for standard installation (LiveCD). It is now installed in the computer not running from the LiveCD. I could try the custom installation to see if performance is improved.

yapatel, 2017-06-14:
Can you please run "lspci -vvv" in a terminal and paste the output here?

cpalvarezb, 2017-06-14:
00:00.0 Host bridge: Intel Corporation 4 Series Chipset DRAM Controller (rev 03)
	Subsystem: Dell Device 027f
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ >SERR- <PERR- INTx-
	Latency: 0
	Capabilities: <access denied>

00:01.0 PCI bridge: Intel Corporation 4 Series Chipset PCI Express Root Port (rev 03) (prog-if 00 [Normal decode])
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 0000d000-0000dfff
	Memory behind bridge: fe900000-feafffff
	Prefetchable memory behind bridge: 00000000d0000000-00000000dfffffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA+ MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: <access denied>
	Kernel driver in use: pcieport

00:03.0 Communication controller: Intel Corporation 4 Series Chipset HECI Controller (rev 03)
	Subsystem: Dell Device 027f
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 50
	Region 0: Memory at f0400800 (64-bit, non-prefetchable) [size=16]
	Capabilities: <access denied>
	Kernel driver in use: mei

00:03.2 IDE interface: Intel Corporation 4 Series Chipset PT IDER Controller (rev 03) (prog-if 85 [Master SecO PriO])
	Subsystem: Dell Device 027f
	Control: I/O+ Mem- BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz+ UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin C routed to IRQ 18
	Region 0: I/O ports at fe80 [size=8]
	Region 1: I/O ports at fe90 [size=4]
	Region 2: I/O ports at fea0 [size=8]
	Region 3: I/O ports at feb0 [size=4]
	Region 4: I/O ports at fef0 [size=16]
	Capabilities: <access denied>
	Kernel driver in use: ata_generic

00:03.3 Serial controller: Intel Corporation 4 Series Chipset Serial KT Controller (rev 03) (prog-if 02 [16550])
	Subsystem: Dell Device 027f
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz+ UDF- FastB2B+ ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin B routed to IRQ 17
	Region 0: I/O ports at ec98 [size=8]
	Region 1: Memory at febd8000 (32-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: serial

00:19.0 Ethernet controller: Intel Corporation 82567LM-3 Gigabit Network Connection (rev 02)
	Subsystem: Dell Device 027f
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 48
	Region 0: Memory at febe0000 (32-bit, non-prefetchable) [size=128K]
	Region 1: Memory at febd9000 (32-bit, non-prefetchable) [size=4K]
	Region 2: I/O ports at ecc0 [size=32]
	Capabilities: <access denied>
	Kernel driver in use: e1000e

00:1a.0 USB controller: Intel Corporation 82801JD/DO (ICH10 Family) USB UHCI Controller #4 (rev 02) (prog-if 00 [UHCI])
	Subsystem: Dell Device 027f
[... 147 lines cut ...]
	Region 4: I/O ports at fec0 [size=32]
	Region 5: Memory at f0400000 (32-bit, non-prefetchable) [size=2K]
	Capabilities: <access denied>
	Kernel driver in use: ahci

00:1f.3 SMBus: Intel Corporation 82801JD/DO (ICH10 Family) SMBus Controller (rev 02)
	Subsystem: Dell Device 027f
	Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin C routed to IRQ 9
	Region 0: Memory at febdb000 (64-bit, non-prefetchable) [size=256]
	Region 4: I/O ports at ece0 [size=32]

01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RV620 LE [Radeon HD 3450] (prog-if 00 [VGA controller])
	Subsystem: Dell OptiPlex 980
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 51
	Region 0: Memory at d0000000 (64-bit, prefetchable) [size=256M]
	Region 2: Memory at fe9f0000 (64-bit, non-prefetchable) [size=64K]
	Region 4: I/O ports at dc00 [size=256]
	Expansion ROM at fea00000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: radeon

yapatel, 2017-06-15:
It looks like you have all the right hardware and drivers. Can you paste the output of the following command:

`cat /proc/cpuinfo`

cpalvarezb, 2017-06-15:
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 23
model name	: Intel(R) Core(TM)2 Duo CPU     E8400  @ 3.00GHz
stepping	: 10
microcode	: 0xa07
cpu MHz		: 2992.370
cache size	: 6144 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fdiv_bug	: no
hlt_bug		: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx lm constant_tsc arch_perfmon pebs bts aperfmperf pni dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm sse4_1 xsave lahf_lm dtherm tpr_shadow vnmi flexpriority
bogomips	: 5984.74
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 23
model name	: Intel(R) Core(TM)2 Duo CPU     E8400  @ 3.00GHz
stepping	: 10
microcode	: 0xa07
cpu MHz		: 2992.370
cache size	: 6144 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fdiv_bug	: no
hlt_bug		: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx lm constant_tsc arch_perfmon pebs bts aperfmperf pni dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm sse4_1 xsave lahf_lm dtherm tpr_shadow vnmi flexpriority
bogomips	: 5984.74
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

yapatel, 2017-06-15:
I recommend trying the custom configuration as described [here](http://rtxi.org/install/) - the generic configuration we provide with the LiveCD is not optimal for your processor.

cpalvarezb, 2017-06-16:
Hi,

I run the custom installation but got stuck after running ./install_rt_kernel.sh

The process stops with an error and the window to edit the kernel configuration doesn't pop up. Here is the whole  output:

-----> Setting up variables.
-----> Environment configuration complete.
-----> Downloading Linux kernel.
--2017-06-16 12:51:10--  https://www.kernel.org/pub/linux/kernel/v3.x/linux-3.8.13.tar.bz2
Resolving www.kernel.org (www.kernel.org)... 147.75.196.57, 2604:1380:1:3600::3
Connecting to www.kernel.org (www.kernel.org)|147.75.196.57|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 84584333 (81M) [application/x-bzip2]
Saving to: ‘linux-3.8.13.tar.bz2’

100%[======================================>] 84,584,333  12.0MB/s   in 11s    

2017-06-16 12:51:22 (7.22 MB/s) - ‘linux-3.8.13.tar.bz2’ saved [84584333/84584333]

-----> Downloading Xenomai.
--2017-06-16 12:51:35--  https://xenomai.org/downloads/xenomai/stable/xenomai-2.6.4.tar.bz2
Resolving xenomai.org (xenomai.org)... 212.83.143.187
Connecting to xenomai.org (xenomai.org)|212.83.143.187|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 21193853 (20M) [application/x-bzip2]
Saving to: ‘xenomai-2.6.4.tar.bz2’

100%[======================================>] 21,193,853  7.24MB/s   in 2.8s   

2017-06-16 12:51:38 (7.24 MB/s) - ‘xenomai-2.6.4.tar.bz2’ saved [21193853/21193853]

-----> Downloads complete.
-----> Patching kernel.
checking file arch/x86/Kconfig
checking file arch/x86/Kconfig.debug
checking file arch/x86/include/asm/apic.h
checking file arch/x86/include/asm/apicdef.h
checking file arch/x86/include/asm/debugreg.h
checking file arch/x86/include/asm/desc.h
checking file arch/x86/include/asm/fpu-internal.h
checking file arch/x86/include/asm/hw_irq.h
checking file arch/x86/include/asm/i8259.h
checking file arch/x86/include/asm/ipi.h
checking file arch/x86/include/asm/ipipe.h
checking file arch/x86/include/asm/ipipe_32.h
checking file arch/x86/include/asm/ipipe_64.h
checking file arch/x86/include/asm/ipipe_base.h
checking file arch/x86/include/asm/irq_vectors.h
checking file arch/x86/include/asm/irqflags.h
checking file arch/x86/include/asm/mmu_context.h
checking file arch/x86/include/asm/page_64_types.h
checking file arch/x86/include/asm/processor.h
checking file arch/x86/include/asm/special_insns.h
checking file arch/x86/include/asm/switch_to.h
checking file arch/x86/include/asm/thread_info.h
checking file arch/x86/include/asm/traps.h
checking file arch/x86/include/asm/tsc.h
checking file arch/x86/kernel/Makefile
checking file arch/x86/kernel/apic/apic.c
checking file arch/x86/kernel/apic/apic_flat_64.c
checking file arch/x86/kernel/apic/io_apic.c
checking file arch/x86/kernel/apic/ipi.c
checking file arch/x86/kernel/apic/x2apic_cluster.c
checking file arch/x86/kernel/apic/x2apic_phys.c
checking file arch/x86/kernel/cpu/common.c
checking file arch/x86/kernel/cpu/mtrr/cyrix.c
checking file arch/x86/kernel/cpu/mtrr/generic.c
checking file arch/x86/kernel/dumpstack_32.c
checking file arch/x86/kernel/dumpstack_64.c
[... 125 lines cut ...]
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
using config: '/boot/config-4.4.0-31-generic'
8250_fintek config not found!!
input_leds config not found!!
irqbypass config not found!!
mei_me config not found!!
amdkfd config not found!!
snd_hda_core config not found!!
snd_hda_codec_generic config not found!!
*
* Restart config...
*
*
* General setup
*
Prompt for development and/or incomplete code/drivers (EXPERIMENTAL) [N/y/?] (NEW) aborted!

Console input/output is redirected. Run 'make oldconfig' to update configuration.

make[1]: *** [localmodconfig] Error 1
make: *** [localmodconfig] Error 2

yapatel, 2017-06-20:
Hi Claudia - it would be best if I could take a look at your system. Can you please install [TeamViewer](https://www.teamviewer.com/en/download/linux/) and send the ID/password to yogi at rtxi.org? 

Please do not post them to GitHub!

closed 2017-06-24 by cpalvarezb