RTXI/rtxi #113: Analogy driver problem after running rtxi install script

wosen7, 2017-06-14:
## Basic Information: 
**RTXI Version:** 2.0
**Installed from Live CD?:** No, cloned from git repos
**PC:** Dell OptiPlex-7040 
**kernel:** 3.18.20-xenomai-2.6-2.6.5

## Error Description: 

**What happened?**
After running the install_rtxi script I get the following message:
```
-----> RTXI intallation successful.
-----> Putting things into place.
-----> Load analogy driver with systemd
Job for rtxi_load_analogy.service failed because the control process exited with error code. See "systemctl status rtxi_load_analogy.service" and "journalctl -xe" for details.
```
When running the suggested commands I get:
**systemctl status rtxi_load_analogy.service**
```
rtxi_load_analogy.service - Load analogy_ni_pcimio at boot
   Loaded: loaded (/etc/systemd/system/rtxi_load_analogy.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-06-14 18:08:52 EDT; 7min ago
  Process: 29968 ExecStart=/bin/bash -c if [[ `uname -r` =~ "xenomai" ]]; then /usr/xenomai/sbin/analogy_config analogy0 analogy_ni_pcimio; fi (code=exited, status=137)
 Main PID: 29968 (code=exited, status=137)

Jun 14 18:08:52 david-OptiPlex-7040 systemd[1]: Starting Load analogy_ni_pcimio at boot...
Jun 14 18:08:52 david-OptiPlex-7040 bash[29968]: /bin/bash: line 1: 29971 Killed                  /usr/xenomai/sbin/analogy_config analogy0 analogy_ni_pcimio
Jun 14 18:08:52 david-OptiPlex-7040 systemd[1]: rtxi_load_analogy.service: Main process exited, code=exited, status=137/n/a
Jun 14 18:08:52 david-OptiPlex-7040 systemd[1]: Failed to start Load analogy_ni_pcimio at boot.
Jun 14 18:08:52 david-OptiPlex-7040 systemd[1]: rtxi_load_analogy.service: Unit entered failed state.
Jun 14 18:08:52 david-OptiPlex-7040 systemd[1]: rtxi_load_analogy.service: Failed with result 'exit-code'.
```

**journalctl -xe**
```
Jun 14 18:18:29 david-OptiPlex-7040 kernel: RIP  [<ffffffff816ca3f1>] rthal_strncpy_from_user+0x11/0x30
Jun 14 18:18:29 david-OptiPlex-7040 kernel:  RSP <ffff8803ac063e68>
Jun 14 18:18:29 david-OptiPlex-7040 kernel: CR2: 00007ffe12e85f38
Jun 14 18:18:29 david-OptiPlex-7040 kernel: ---[ end trace 1151ecf951b566f0 ]---
Jun 14 18:18:29 david-OptiPlex-7040 bash[5464]: /bin/bash: line 1:  5467 Killed                  /usr/xenomai/sbin/analogy_config analogy0 analogy_ni_pcimio
Jun 14 18:18:29 david-OptiPlex-7040 systemd[1]: rtxi_load_analogy.service: Main process exited, code=exited, status=137/n/a
Jun 14 18:18:29 david-OptiPlex-7040 systemd[1]: Failed to start Load analogy_ni_pcimio at boot.
-- Subject: Unit rtxi_load_analogy.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit rtxi_load_analogy.service has failed.
-- 
-- The result is failed.
Jun 14 18:18:29 david-OptiPlex-7040 systemd[1]: rtxi_load_analogy.service: Unit entered failed state.
Jun 14 18:18:29 david-OptiPlex-7040 systemd[1]: rtxi_load_analogy.service: Failed with result 'exit-code'.
```

I am stuck. I attempted to recompile the kernel and RTXI but error persists.

Any idea on how to proceed from here? Please let me know what additional information about my system could be helpful.

Thank you very much!
David

yapatel, 2017-06-15:
Let's try to load the driver manually.

1. Please run the following in the command line:

`sudo /usr/xenomai/sbin/analogy_config analogy0 analogy_ni_pcimio`

If that does not work...

2. Please type `lspci -vvv` at the command line and find the entry that has the information for your DAQ.  It should look something like this:

`04:01.0 Unassigned class [ff00]: National Instruments PCI-6259`
`Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV+ VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-`
`Status: Cap- 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-`
`Latency: 64, Cache Line Size: 64 bytes`
`Interrupt: pin A routed to IRQ 17`
`Region 0: Memory at dfafe000 (32-bit, non-prefetchable) [size=4K]`
`Region 1: Memory at dfaff000 (32-bit, non-prefetchable) [size=4K]`
`Kernel driver in use: analogy_mite`

The line of interest is:

`04:01.0 Unassigned class [ff00]: National Instruments PCI-6259`

This tells us the bus address for the DAQ. In this case, the PCI bus address is 0x4 and the specific slot is 1. Once you have this information, please try:

`sudo /usr/xenomai/sbin/analogy_config analogy0 analogy_ni_pcimio <bus>,<slot>` (in this case it would be 0x4,1).

wosen7, 2017-06-22:
Hi Yogi, thanks for advice.

1. the console output is only: `Killed`

2. `lspci -vvv` gives:
```
05:00.0 Unassigned class [ff00]: National Instruments PCIe-6251
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 32
	Interrupt: pin A routed to IRQ 18
	Region 0: Memory at f7001000 (32-bit, non-prefetchable) [size=4K]
	Region 1: Memory at f7000000 (32-bit, non-prefetchable) [size=4K]
	Kernel driver in use: ni_pcimio
	Kernel modules: ni_pcimio
```
Thus I ran:
`sudo /usr/xenomai/sbin/analogy_config analogy0 analogy_ni_pcimio 0x5, 0`
Again the output on the console is: `Killed`

Inescaetano91, 2017-09-13:
I have the same problem but I think it was because I don't have the acquisition board connected to my system. What acquisition board are you using ?

yapatel, 2017-09-13:
Yes, analogy_config will throw an error if there is no compatible DAQ on board. Compatible DAQs can be found [here](http://rtxi.org/docs/manual/#daqs).

closed 2017-09-13 by yapatel