RTXI/rtxi #114: Does Anyone Run RTXI On 16.04 Ubuntu and/or newer hardware (Skylake/KabyLake)?

jagalli, 2017-06-15:
Hello,

I'm looking to order a system for one of our clients, but it looks like the base RTXI image is only capable of running on 14.04 and limited hardware because of the kernel (broadwell/haswell chipsets or older).

As a University IT group, we'd prefer our users have a system that is under a 3 year plus warranty, but it's been difficult to find older hardware that fits the bill.  Has anyone successfully got this working on 16.04?  I see the package for RTXI is available there.  Or are there a slew of issues?

Ideally we'd want to order something like an Optiplex 7050 with proper video card.  Also has anyone had good success with certain video card models from the Ubuntu recommended chart?  I notice they suggest for RTXI to use an AMD Radeon card with open source drivers

Thank you, any help is greatly appreciated

- John Galli
University of Wisconsin

yapatel, 2017-06-20:
Yes, we have RTXI running on 16.04. We are in the process of testing it on the new Skylake processors. It will likely take 3-4 weeks.  For video cards, newer Ubuntu releases provide more support for Nvidia cards - so I suggest purchasing that. Will keep you posted on the testing once complete.

jagalli, 2017-06-20:
Hello Yogi,

Thanks for your reply, that’s promising that it’s running on 16.04 (and I assume an updated kernel).  We’ve been trying to get RTXI running on several Dell systems we have here with the following configurations:

Dell Optiplex 7050 (KabyLake i5)
Radeon RV610 and RV635 (have tried each separately)
16 GB RAM (2x8gb dual channel)
SSD
Ubuntu 14.04 ; 4.1.18 Kernel

Dell Optiplex 7010 (IvyBridge i5) (have tried each separately)
Radeon RV610 and RV635
8 GB RAM (2x4gb dual channel)
Live CD (RTXI) 3.8.1 Kernel

Both systems seem to be getting slightly higher latency numbers than the benchmarks on the RTXI site, and we’re getting a large spike every 10-15 minutes (on average).  We’re not certain if this is a chipset issue or some feature of the Dell BIOS (which we’ve stripped down to have most speed step and power saving features off) or something else.  It does run a little better on the 3.8.1 Live CD (on the older system), but both are still less than ideal.  I've attached our benchmark data, any ideas what might be causing the issue?
[image]
  

John

[test_rt_histdata.txt](https://github.com/RTXI/rtxi/files/1088444/test_rt_histdata.txt)

jagalli, 2017-06-20:
Also do you know of any specific computers that are orderable with warranties which are known to work close to benchmark results for RTXI?  Given the request for older hardware, we've had difficulties finding supported systems

yapatel, 2017-06-24:
Can you try removing the graphics card and running off of on-board graphics? The kernel should have the default intel drivers for on-board graphics. This should help identify if the graphics card is the issue. 

Regarding the computers, we are running a few tests here. I can let you know as soon as some of those are validated (again a few weeks).

yapatel, 2017-07-06:
We now have a stable 16.04 version compatible with the Skylake and Kaby Lake processors. If you are still interested in setting this up, please checkout the [xenial](https://github.com/RTXI/rtxi/tree/xenial) branch and run the appropriate installation scripts in the `scripts` folder. The xenial branch will be merged to master in the coming days. Any feedback you provide is greatly appreciated!

closed 2017-09-13 by yapatel