RTXI/rtxi #115: Installing the real-time kernel error

philippsasse, 2017-06-20:
## Basic Information: 
RTXI Version: 2.0
Installed from Live CD?:  No, cloned from git repos
PC: HP Z400 XEON CPU E5-1630 v3 

## Error Description: 

**What happened?**
Fresh installation onto  Ubuntu GNOME 14.04.5 following exactly the website documentation terminated at the step ./install_rt_kernel.sh with fhe following errors:

## Error Message: 
```
checking file mm/vmalloc.c
HOSTCC scripts/basic/fixdep
HOSTCC scripts/kconfig/conf.o
SHIPPED scripts/kconfig/zconf.tab.c
SHIPPED scripts/kconfig/zconf.lex.c
SHIPPED scripts/kconfig/zconf.hash.c
HOSTCC scripts/kconfig/zconf.tab.o
HOSTLD scripts/kconfig/conf
using config: '/boot/config-4.4.0-31-generic'
intel_rapl config not found!!
intel_powerclamp config not found!!
comedi_pci config not found!!
snd_hda_core config not found!!
crc32_pclmul config not found!!
comedi_8255 config not found!!
irqbypass config not found!!
8250_fintek config not found!!
snd_hda_codec_generic config not found!!
input_leds config not found!!
fjes config not found!!
x86_pkg_temp_thermal config not found!!
module ablk_helper did not have configs
CONFIG_CRYPTO_ABLK_HELPER_X86
.config:6864:warning: symbol value 'm' invalid for
COMEDI_PCI_DRIVERS
*
* Restart config...
*
*
* General setup
*
Prompt for development and/or incomplete code/drivers (EXPERIMENTAL) [N/y/?] (NEW) aborted!
Console input/output is redirected. Run 'make oldconfig' to update configuration.
make[1]: *** [localmodconfig] Error 1
make: *** [localmodconfig] Error 2
rtxiuser@rtxi-HP-Z440-Workstation:~/rtxi/scripts$ make
oldconfig
make: *** No rule to make target `oldconfig'. Stop.
```

yapatel, 2017-06-24:
It would be best if I could take a look at your system. Can you please install TeamViewer and send the ID/password to yogi at rtxi.org? Please do not post this information to GitHub!

yapatel, 2017-06-29:
Please see [solution here](https://github.com/RTXI/rtxi/issues/116#issuecomment-311990300)

closed 2017-06-29 by yapatel