RTXI/rtxi #116: Errors in RT kernel installation

jadwn, 2017-06-29:
## Basic Information: 
**RTXI Version: 2.0
**Installed from Live CD?: No, from github repository

## Error Description: 

**What happened?**
I followed the instructions for a custom installation. Installing the real-time kernel caused the below errors.

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```
checking file mm/vmalloc.c
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
using config: '/boot/config-4.4.0-83-generic'
intel_lpss_acpi config not found!!
fjes config not found!!
intel_lpss config not found!!
intel_lpss_pci config not found!!
crc32_pclmul config not found!!
idma64 config not found!!
btbcm config not found!!
acpi_als config not found!!
comedi_pci config not found!!
snd_hda_codec_generic config not found!!
8250_fintek config not found!!
i915_bpo config not found!!
tpm_crb config not found!!
snd_hda_core config not found!!
crct10dif_pclmul config not found!!
intel_rapl config not found!!
x86_pkg_temp_thermal config not found!!
comedi_8255 config not found!!
mei_me config not found!!
input_leds config not found!!
btqca config not found!!
intel_powerclamp config not found!!
irqbypass config not found!!
btintel config not found!!
WARNING: DMA_VIRTUAL_CHANNELS is required, but nothing in the
  current config selects it.
WARNING: DMA_VIRTUAL_CHANNELS is required, but nothing in the
  current config selects it.
module ablk_helper did not have configs CONFIG_CRYPTO_ABLK_HELPER_X86
.config:6668:warning: symbol value 'm' invalid for COMEDI_PCI_DRIVERS
*
* Restart config...
*
*
* General setup
*
Prompt for development and/or incomplete code/drivers (EXPERIMENTAL) [N/y/?] (NEW) aborted!

Console input/output is redirected. Run 'make oldconfig' to update configuration.

make[1]: *** [localmodconfig] Error 1
make: *** [localmodconfig] Error 2

```

Thanks,
Jad

yapatel, 2017-06-29:
Please change the following line in install_rt_kernel.sh:

`yes "" | make localmodconfig`

to

`yes "" | make oldconfig`

and rerun the script.

jadwn, 2017-06-29:
The kernel was installed, but hangs up when booting into it. It hangs at:
```
Loading initial ramdisk ...
```
Is there a solution to that?

closed 2017-09-13 by yapatel