RTXI/rtxi #119: RTXI 2.0 amd64 image installed, but xenomai test failed

oPiZiL, 2017-10-27:
RTXI 2.0 amd64
## Basic Information: 
**RTXI Version:RTXI 2.0 amd64
**Installed from Live CD?:RTXI 2.0 amd64

## Error Description: 

**What happened?**
./test_rt_kernel.sh failed

**What was supposed to happen?**
/dev/rtheap missing

**What steps will trigger the error?**

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```
xenomai : /dev/rtheap missing
```

yapatel, 2017-10-27:
Please make sure you are booted into the RT kernel. You can check which kernel you are booted into by typing `uname -r`

oPiZiL, 2017-10-30:
`uname -r`
`3.16.0-77-generic`
So, how could I boot into RT kernel

oPiZiL, 2017-10-30:
I modified 

> /etc/default/grub
`GRUB_HIDDEN_TIMEOUT=0`-->`GRUB_HIDDEN_TIMEOUT=10`

then I restart the system. The screen goes grey, So I press ESC. I selected xenomai mennuentry, BUT I can't go into graphics desktop.

yapatel, 2017-11-01:
It sounds like you may be missing the right graphics driver for your card. Are you using a dedicated graphics card or on board graphics?

oPiZiL, 2017-11-01:
It's an onboard graphic card.

 one more question:

If I upgrade the Ubuntu 14.04 to 16.04, would it work in RT kernel?

yapatel, 2017-11-01:
Typically the OS and the kernel are not dependent upon each other. That being said, there could be kernel dependencies that cause issues.  

If you have 14.04 working, then there is no significant benefit to jumping to 16.x. Both are LTS versions and function for RTXI.  

Yogi

yapatel, 2017-11-20:
has this issue been resolved?

oPiZiL, 2017-11-22:
RT kernel  works after upgrade

closed 2021-04-01 by fusge