RTXI/rtxi #120: -lrtdsp library problem, Upgrade to 2.1 dependency problem, Xenomai settings for Xenial problem

tjr1, 2017-10-27:
Hi,

TL;DR: Please skip to #3 below and help with Xenomai 3 settings.

I've been trying to get a fully working version of RTXI for the past few weeks.  I've tried it three ways and have run into road blocks in each case.  I'll describe the three ways I've gone, and if you'd please suggest which of the 3 is most easily solved.

1.  I intalled the live CD with ubuntu 14.04 and RTXI 2.0.  It works, except the FIR and IIR modules fail to build.  They cannot find the -lrtdsp library.  I can see that you edited these modules in December 2016.  Maybe RTXI 2.0 is no longer compatible with the recent versino of the module.

2.  So then I then tried upgrading to 2.1  following the instructions on RTXI.org.  While installing the dependencies, it failed with 3 problems.  
The first I was able to solve by downgrading libssl:
sudo apt-get install libssl1.0.0=1.0.1f-1ubuntu2
But the other two are illusive.  It says: 
The following packages have unmet dependencies:
 qt5-default : Depends: qtbase5-dev but it is not going to be installed
 qttools5-dev : Depends: qtbase5-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
I tried to install/downgrade the dependencies to make it work, but I couldnt' solve it.  

3.  So that's when I decied to to go for broke and compile the Xenial branch.  I installed Ubuntu 16 from USB and followed the install steps.  I managed to make it all of the way through installation.  The hardest part was the Xenomai configuration.  When I restarted and selected the Xenomai kernel in Grub, it just hung and didn't post.  Would it do this if my Xenomai configuration was wrong?  I tried to disable/enable according to the instructions, but the instructions were written for Xenomai 2, not 3, and it is different.  I tried deleting the Xenomai folder and then reinstalling it with different settings a couple times, but it didn't work.  

Thanks for your help.  Truly grateful for RTXI--it is a huge part of our lab.
Best,
Tom

## Basic Information: 
**RTXI Version:** 
**Installed from Live CD?:**

## Error Description: 

**What happened?**

**What was supposed to happen?**

**What steps will trigger the error?**

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```

```

yapatel, 2017-10-29:
Hi Tom -

Thanks for the thorough description of what you have tried so far. Can you please provide the output to the following command:

$ cat /proc/cpuinfo

Also, can you send the .config file that is in the “build” folder that is created by the installation script? It should be in /opt.

tjr1, 2017-10-30:
Hi Yogi,

Many many thanks for your help (and for developing RTXI).  I attached my cpu info.  It is an i5-3570 (ivy bridge).  I also attached the requested config.log file.  

I tried reinstalling Ubuntu and and Xenomai from scratch again and this time I took good notes.  Of most important note: It wouldn't allow me to disable ACPI for the processor nor to disable cpu idle PM support.  They were fixed at _*_ no matter what I did.  I was able to disable suspend to ram, hibernation, cpu frequency, CpuIdle driver for Intel Processors, and comedi.  

[config.log](https://github.com/RTXI/rtxi/files/1428950/config.log)

[processor_info.txt](https://github.com/RTXI/rtxi/files/1428938/processor_info.txt)

Thank you,
Tom

yapatel, 2017-11-01:
I'll have to try to take a look at your system directly. Can you install teamviewer and email the info to yogi at rtxi dot org?

tjr1, 2017-11-01:
Hi,
That would be amazing.  I will install teamviewer and email you the info in the morning.

closed 2021-04-01 by fusge