RTXI/rtxi #121: Unable to boot the Real-Time kernel (3.8.13-xenomai-2.6.3)

URSUroman, 2017-11-07:
## Basic Information: 
**RTXI Version: 76a031b5ed6c2094b9febd1cf514e4d2d1774bf3** 
**Installed from Live CD?: No. Downloaded from GitHub and installed with the script install_rt_kernel.sh**

## Error Description: 

**What happened?**
First, I installed the Ubuntu GNOME 14.04.5 i386.
I booted Ubuntu GNOME and within it, I run the script install_rt_kernel.sh. After install_rt_kernel.sh           successfully completed execution, I shut-down the computer and powered it on anew. In the GRUB menu I chose to   boot the 3.8.13-xenomai-2.6.3 system. 
After that, the screen displayed:
"Loading init ramdisk" (it is normal. The GNOME boot-up gives the same at this stage of the boot)

After that, the screen turned black displaying:
"tcs: Fast TCS calibration failed"

Shortly after the screen turned dark and blank and nothing changed after that. The 3.8.13-xenomai-2.6.3 system doen't boot.

Moreover:
lspci gives for the graphics card:
"01:00.0 VGA compatible controller: NVIDIA Corporation Device 1c81 (rev a1)"

lspci executed on the same computer but with ubuntu 16 also gave for the graphics card
Kernel modules: Nouveau, other modules...

I therefore assume, that my graphics card should be supported by xenomai (since the card is supported by Noveau driver). I also moved my graphics card to an other pci slot but the problem with 3.8.13-xenomai-2.6.3 persisted.
I also recompiled the 3.8.13-xenomai-2.6.3 system (by re-executing the script install_rt_kernel.sh) but the problem persisted.
Finally, while selecting 3.8.13-xenomai-2.6.3 in GRUB menu, I also pressed 'e' and added 'nomodeset' to the line 'linux /boot' but at the end I also got a blank dark screen with no boot up of the 3.8.13-xenomai-2.6.3 system.

Do you have an idea what I can do to boot the 3.8.13-xenomai-2.6.3 system?

Sincere regards,

Roman

yapatel, 2017-11-08:
Does your computer have an on board graphics port? If so, have you tried that?

Are you able to boot into the vanilla kernel (i.e. kernel without Xenomai)?

yapatel, 2017-11-20:
Before going too deep - have you tried the live cd? http://rtxi.org/install/ for 14.04?

URSUroman, 2017-11-22:
I think the problem was that my hardware was too recent: I have an "i7-7700K" with an integrated "Intel HD Graphics 630" card. The graphics driver in kernel 3.8.13 probably doesn't support the HD Graphics 630 card. My assumption is that kernel 3.8.13 has the driver i915 for the intel's integrated graphic card however the new HD Graphics 630 card needs a newer version of that driver: i915_bpo. The kernel 3.8.13 doesn't have this newer driver. That's probably why I could not get to the login screen.

I tried installing the system (linux-3.8.13-xenomai-2.6.3) on a Core2 vPro and finally got a working system. The latency test provides maximal latencies not exceeding 20 microsec.

I also remarked, you put rtxi on a newer kernel 4.9.24. I tried installing (linux-4.9.24-xenomai-3.0.5) on the i7-7700K and rtxi installed normally. However the latency tests constantly provide max latencies of 120 microsecs with worst latencies around 1ms. 

I close this topic since it was about (linux-3.8.13-xenomai-2.6.3) and open a new topic for the latency problem with linux4.9.24-xenomai-3.0.5.

closed 2017-11-22 by URSUroman