RTXI/rtxi #123: Errors 1 & 2 installing RT kernel

jenniNelson, 2017-11-15:
**RTXI Version:** 2.1
**Installed from Live CD?:** No

## Error Description: 

When running install_rt_kernel.sh, after the configuration settings I keep getting the below errors.
I tried fixing it with one of the other fixes on here (going to the file and swapping oldconfig with localmodconfig) but the change had already been made, I think. Tried a few versions on that which threw errors before the configuration pop-up. It's reverted now, and still not working.

I'm running this in Ubuntu. I tried to install this once before but got stalled on a still-unknown issue before getting to the configuration pop-up. I ran the uninstall script, erased the rtxi file and restarted. (That error didn't show up this time around.)

## Error Message: 
```
====== making target debian/stamp/build/kernel [new prereqs: vars]======
This is kernel package version 13.018.
restore_upstream_debianization
test ! -f scripts/package/builddeb.kpkg-dist ||	mv -f scripts/package/builddeb.kpkg-dist scripts/package/builddeb
test ! -f scripts/package/Makefile.kpkg-dist ||	mv -f scripts/package/Makefile.kpkg-dist scripts/package/Makefile
/usr/bin/make -j8 EXTRAVERSION=-xenomai-3.0.5  ARCH=x86_64 \
		     bzImage
make[1]: Entering directory '/opt/linux-4.9.24'
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/asn1_compiler
  CC      scripts/mod/empty.o
  HOSTCC  scripts/sortextable
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/sign-file
  CHK     include/generated/utsrelease.h
scripts/sign-file.c:25:30: fatal error: openssl/opensslv.h: No such file or directory
compilation terminated.
scripts/Makefile.host:107: recipe for target 'scripts/sign-file' failed
make[2]: *** [scripts/sign-file] Error 1
make[2]: *** Waiting for unfinished jobs....
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/selinux/mdp/mdp
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/sumversion.o
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  HOSTLD  scripts/mod/modpost
Makefile:560: recipe for target 'scripts' failed
make[1]: *** [scripts] Error 2
make[1]: Leaving directory '/opt/linux-4.9.24'
debian/ruleset/targets/common.mk:295: recipe for target 'debian/stamp/build/kernel' failed
make: *** [debian/stamp/build/kernel] Error 2

```

yapatel, 2017-11-16:
which version of ubuntu and linux kernel are you attempting to build on?

jenniNelson, 2017-11-20:
The most recent Ubuntu, 16.04.03. It says my kernel is 4.10.0-38-generic, which I think is the default for Ubuntu. I just barely installed this OS on an old computer

yapatel, 2017-11-29:
Can you try running the custom installation scripts in the master branch on your newly installed OS? See here for more information: http://rtxi.org/install/

 We are still updating the documentation for 16.04 so there may be some items missing. If you have trouble, email yogi at rtxi dot org and I can take a look remotely.

jenniNelson, 2017-11-29:
I've actually gone ahead and started over with the easier standard version of installation, from a pre-installed linux image. (I don't really need perfect performance for my purposes) But thank you anyways!

yapatel, 2017-11-29:
No problem - let us know if something else comes up!

closed 2017-11-29 by yapatel