RTXI/rtxi #124: Too long latencies with latency test

URSUroman, 2017-11-22:
## Basic Information: 
**RTXI Version:*2afb5e58ff94be7bc3164ba26b203ee90c17b554* 
**Installed from Live CD?:*No, custom installation*

## Error Description: 

I custom installed rtxi on an  i7-7700K.
When running the latency test (RT period like in http://rtxi.org/stats/ i.e. 10kHz), I get typically:
RTH|----lat min|----lat avg|----lat max|-overrun|---msw|---lat best|--lat worst
...
RTD|      0.710|      2.346|    121.659 |    3245|     0|      0.514|   1276.562
RTD|      0.675|      2.362|    131.398 |    3261|     0|      0.514|   1276.562
RTD|      0.704|      2.381|    356.757 |    3279|     0|      0.514|   1276.562
RTD|      0.551|      2.406|    687.567 |    3298|     0|      0.514|   1276.562
RTD|      0.673|      2.491|   1107.889|    3322|     0|      0.514|   1276.562
RTD|      0.703|      2.488|   1333.165|    3348|     0|      0.514|   1333.165
RTD|      0.558|      2.337|    130.778 |    3362|     0|      0.514|   1333.165
...

lat max is always around 120 microsec. 
Sometimes, lat max is around 1200 microsec (this happens every 30-40 seconds).

If I understand it well, lat max should not exceed 20-30 microsecs.

When configuring the kernel, in menuconfig, I don't have the same options as at point 6 in http://rtxi.org/install/custom. This is probably because the doc was not yet adapted for the newer kernel. When trying to configure the kernel as consistent as possible with what is presented at point 6 of http://rtxi.org/install/, I realise that I can't disable the option: 
--> Power Management and ACPI options 
     --> ACPI Support
           --> Processor (disable) 
unless I disable the whole ACPI support.  Since there is a warning in menuconfig about ACPI being able to cause problems with xenomai, I disabled ACPI Support fully.

However, there is a second warning in menuconfig about CONFIG_MIGRATION. I can't get rid of it. I edited the .config file after menuconfig finished and before launching 
fakeroot make-kpkg \
	--initrd \
	--append-to-version=-xenomai-$xenomai_version \
	--revision $(date +%Y%m%d) \
	kernel-image kernel-headers modules
I replaced "CONFIG_MIGRATION=y" with "#CONFIG_MIGRATION is not set" but after launching fakeroot make-kpkg \....., .config contains again "CONFIG_MIGRATION=y" as if something reset my change in .config.

Do you know what can cause max latencies of 120 microsec with bursts of 1200 microsecs? 
Is there a way to get rid of CONFIG_MIGRATION (in case it is the problem)? How can I edit .config so that the changes are taken over by the subsequent make-kpkg command ?

Best regards,
Roman

URSUroman, 2017-12-15:
Hello,

I think that the problem with too long latencies is due to the fact that I disabled the whole ACPI support. I must have disabled too much to ensure a proper functionning of the hardware. For example my processor's clock periodically increases its frequency and then decreases it....

I finally managed to disable only the options as presented in http://rtxi.org/install/custom
To be able to disable the option
--> Power Management and ACPI options
--> ACPI Support
--> Processor (disable)

I fist disabled all the other options that needed to be disabled (i.e. ponts 1, 2, 4 and then  point 3).

Now the latencies of the system are similar to those expected: for a real-time frequency of 10kHz, the average latency is 2µs and maximal latencies are  usually around 13 µs and exceptionalyl 20-30 µs.

Best regards,
Roman

Enable ANALOGY drivers.
From the top menu, follow the path:
--> Real-Time sub-system
--> Drivers
--> Analogy

Hit SPACE on all the drivers to have them built into the kernel. A "*" beside the field means that the drivers are set properly. Note that "M" means that the drivers will be built as kernel modules, not built into the kernel itself. "*" is the desired setting here, not "M".

Hit ESC to get back to the top menu from where you started. Be careful to not exit the configuration menu.

Select your processor type.
From the top menu, follow the path:
--> Processor type and features
--> Processor family

Select the option that corresponds to your processor type. If unsure of what processor you have, open another terminal run:

$ cat /proc/cpuinfo

This will display the CPU information needed to correctly select your configuration. Most newish computers with Intel processors usually have an i3, i5, or i7. For them, select Core 2/newer Xeon.

Note: For 32-bit systems, Enable -fstack-protector buffer overflow detection should also be disabled

Disable power management options.
You need to disable some CPU scaling capabilities of the kernel that can otherwise interrupt the real-time framework. From the top menu, follow the path and disable:
--> Power Management and ACPI options
--> Suspend to RAM (disable)
--> Hibernation (disable)
--> ACPI Support
--> Processor (disable)
--> Frequency Scaling (disable)
--> CPU Idle driver for Intel (disable)
--> CPU Idle PM support (disable)

To disable drivers, move the cursor to the item and hit SPACE until the field the left is blank.

Disable COMEDI drivers.

closed 2017-12-15 by URSUroman

yapatel, 2017-12-15:
Thank you for the excellent documentation and reporting of what you did to get it to work. This will be helpful for future users and we will include a reference to it in the documentation.