RTXI/rtxi #126: Possible IRQ Interferences with Ubuntu 16.04 LTS, but not with 14.04 LTS?

Trevor372, 2018-06-20:
We are having problems booting the computer up using Ubunto 16.04 LTS for use with RTXI.  When we turn the computer on, it will stall out on the loading screen.  This occurs intermittently, some days we have issues, other days it seems to work.  We've been searching the web and the RTXI website for answers and we're suspicious about the Graphics Driver we're using. It's a Radeon HD 5000/6000/7350/8350 Series with a Dell Radeon HD 6350 subsystem; we are unsure if this is an approved card . there may also be IRQ issues with the National card (see below)
 
Recently, we've downgraded back to the Ubuntu 14.04 LTS, and the system appears more stable.  We ran the diagnostic that provides IRQ conflict info on both 14.04 and 16.04.  the outputs of these are given below.  Our interpretation is that there are conflicts between the National card and the microprocessor on 16.04 but not on the 14.04.  Is this correct? Is there a solution to the conflicts that would allow us to upgrade to 16.04 with the National card we have? Please advise
 
Thank you for your help
 
Trevor Haas
 
 
 
Diagnostic test for 14.04

TERMINAL RESPONSE TO: $ lspci -v | less

00:00.0 Host bridge: Intel Corporation 2nd Generation Core Processor Family DRAM Controller (rev 09)
        Subsystem: Dell Device 047e
        Flags: bus master, fast devsel, latency 0
        Capabilities: <access denied>

00:16.0 Communication controller: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1
 (rev 04)
        Subsystem: Dell Device 047e
        Flags: bus master, fast devsel, latency 0, IRQ 44
        Memory at e16b0000 (64-bit, non-prefetchable) [size=16]
        Capabilities: <access denied>
        Kernel driver in use: mei

00:16.3 Serial controller: Intel Corporation 6 Series/C200 Series Chipset Family KT Controller (rev 04) (
prog-if 02 [16550])
        Subsystem: Dell Device 047e
        Flags: bus master, 66MHz, fast devsel, latency 0, IRQ 17
        I/O ports at 40a0 [size=8]
        Memory at e1690000 (32-bit, non-prefetchable) [size=4K]
        Capabilities: <access denied>
        Kernel driver in use: serial

00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
        Subsystem: Dell Device 047e
        Flags: bus master, fast devsel, latency 0, IRQ 41
        Memory at e1600000 (32-bit, non-prefetchable) [size=128K]
        Memory at e1680000 (32-bit, non-prefetchable) [size=4K]
        I/O ports at 4040 [size=32]
        Capabilities: <access denied>
        Kernel driver in use: e1000e

00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controlle
r #2 (rev 04) (prog-if 20 [EHCI])
        Subsystem: Dell Device 047e
        Flags: bus master, medium devsel, latency 0, IRQ 16
        Memory at e1670000 (32-bit, non-prefetchable) [size=1K]
        Capabilities: <access denied>
        Kernel driver in use: ehci-pci

00:1b.0 Audio device: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Control
ler (rev 04)
        Subsystem: Dell Device 047e
        Flags: bus master, fast devsel, latency 0, IRQ 45
        Memory at e1660000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: <access denied>
        Kernel driver in use: snd_hda_intel

00:1c.0 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 (rev b4
) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        Capabilities: <access denied>
        Kernel driver in use: pcieport

00:1c.2 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 3 (rev b4) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0
[... 348 lines cut ...]
vendor_id : GenuineIntel
cpu family     : 6
model          : 42
model name     : Intel(R) Core(TM) i5-2500 CPU @ 3.30GHz
stepping  : 7
microcode : 0x14
cpu MHz        : 3292.752
cache size     : 6144 KB
physical id    : 0
siblings  : 4
core id        : 3
cpu cores : 4
apicid         : 6
initial apicid : 6
fpu       : yes
fpu_exception  : yes
cpuid level    : 13
wp        : yes
flags          : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts
bugs      :
bogomips  : 6587.04
clflush size   : 64
cache_alignment     : 64
address sizes  : 36 bits physical, 48 bits virtual
power management:

Trevor372, 2018-06-21:
This is from the initial terminal response:
traverslab@traverslab-OptiPlex-990:~$ cat /etc/grub.conf
cat: /etc/grub.conf: No such file or directory

******************************************************************************************
And also from the home directory:
traverslab@traverslab-OptiPlex-990:/$ cat /etc/grub.conf
cat: /etc/grub.conf: No such file or directory

I did some searching and I was able to find the etc folder but not the grub.conf file.  So perhaps we might have a file missing.  What steps are necessary to fix this? Could this be causing the issues?

yapatel, 2018-06-21:
Apologies - I meant 

`cat /etc/default/grub`

Trevor372, 2018-06-21:
No worries, here is the response from that command:

traverslab@traverslab-OptiPlex-990:~$ cat /etc/default/grub
# If you change this file, run 'update-grub' afterwards to update
# /boot/grub/grub.cfg.
# For full documentation of the options in this file, see:
#   info -f grub -n 'Simple configuration'

GRUB_DEFAULT=0
GRUB_HIDDEN_TIMEOUT=0
GRUB_HIDDEN_TIMEOUT_QUIET=true
GRUB_TIMEOUT=10
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"
GRUB_CMDLINE_LINUX=""

# Uncomment to enable BadRAM filtering, modify to suit your needs
# This works with Linux (no patch required) and with any kernel that obtains
# the memory map information from GRUB (GNU Mach, kernel of FreeBSD ...)
#GRUB_BADRAM="0x01234567,0xfefefefe,0x89abcdef,0xefefefef"

# Uncomment to disable graphical terminal (grub-pc only)
#GRUB_TERMINAL=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command `vbeinfo'
#GRUB_GFXMODE=640x480

# Uncomment if you don't want GRUB to pass "root=UUID=xxx" parameter to Linux
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
#GRUB_DISABLE_RECOVERY="true"

# Uncomment to get a beep at grub start
#GRUB_INIT_TUNE="480 440 1"
traverslab@traverslab-OptiPlex-990:~$ 

*****************************************************************************************************************************
I should mention that these commands are being run from Ubuntu 14.04 LTS, not sure if that helps

Trevor372, 2018-06-26:
I think I may have localized part of the issue.  After doing some searching, it appears that the the computer has more than a few different kernels that it boots into, and because of the GRUB_DEFAULT=0 line it automatically chooses the first one in the menu entry.  If I understand correctly, RTXI only works when it boots into the 3.8.13-xenomai-2.6.4 kernel.  When we had updated to Ubuntu 16.04 I believe it may have added another kernel to the menu entry which is why we would occasionally have bootup problems.  When it would bootup and we were able to log in we would get an error when running rtxi:

0"000.000| BUG in low_init(): [main] Cobalt core not enabled in kernel

Would this mean that the wrong kernel had been booted into?  If so, then the solution to our issues may be to delete the extra, unnecessary kernels.

emelon8, 2018-07-30:
I'm not sure if it's the exact same problem, but I am also getting the error

0"000.000| BUG in low_init(): [main] Cobalt core not enabled in kernel

when I try to start RTXI.

When I first installed RTXI 2.2 a few weeks ago, I found that if I installed the live image with the ethernet cable disconnected, I did not have a problem starting RTXI (and could even reconnect the ethernet cable after installation was complete and be fine). However, if I installed the live image with the updates, or even with the ethernet cable connected (without installing updates), then I would get the "0"000.000| BUG in low_init(): [main] Cobalt core not enabled in kernel" error. So I installed it without the ethernet cable connected and thought everything was fine until today. However, after starting the computer today I found the same error. Perhaps related, I also see that it's trying to get me to install updates again (I haven't installed any as far as I know?). I wonder if it has to do with the updates somehow, even if you don't install them?

Trevor372, 2018-08-01:
The problem you are refering to: 
0"000.000| BUG in low_init(): [main] Cobalt core not enabled in kernel
could be because your computer did not boot into the correct kernel for rtxi.  Type uname -r at the terminal and see if your kernel reads out something like 3.8.13-xenomai-2.6.4-aufs.  If it doesn't have xenomai in the readout then you booted into the wrong kernel.

emelon8, 2018-08-02:
Thank you! I was indeed running the wrong kernel. Any advice on how to change which kernel it boots into? I don't see any grub screen when I start the computer for some reason, it just boots right into the wrong kernel.

yapatel, 2018-08-02:
@emelon8 

> Menu will appear if you press and hold Shift during loading Grub.
> 
> For permanent change you'll need to edit your /etc/default/grub file -- place a "#" symbol at the start of line GRUB_HIDDEN_TIMEOUT=0.
> 
> Save changes and run sudo update-grub to apply changes.

yapatel, 2018-08-02:
@Trevor372 lost track here - was your issue resolved? you may just want to update your grub menu default entry. I recommend keeping the other kernels incase you need to use a vanilla kernel to test drivers/troubleshoot/etc

Trevor372, 2018-08-16:
@yapatel We are still running rtxi on Ubuntu 14.04 LTS and it seems to be working well for our purposes.  We haven't had any issues with the bootup on the older 14.04 like we would sometimes have with 16.04.  For the time being updating to the newer version just doesn't make sense for us since we have it working well and so our issue is more or less resolved.  Thanks.

closed 2018-08-16 by Trevor372