RTXI/rtxi #127: Cobalt core not enabled in kernel

apantazis82, 2018-08-20:
## Basic Information: 
**RTXI Version:** 
2.2
**Installed from Live CD?:**
yes (USB)

## Error Description: 
RTXI app doesn't start,
**What happened?**

In the terminal, I get the error "0"000.000| BUG in low_init(): [main] Cobalt core not enabled in kernel"
When I run "uname -r" (as suggested in other thread) I get "4.15.0-32-generic"

**What was supposed to happen?**
RTXI should run

**What steps will trigger the error?**
executing rtxi

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```
0"000.000| BUG in low_init(): [main] Cobalt core not enabled in kernel
```

yapatel, 2018-08-21:
It appears you’re not booted into the real-time kernel. When the computer starts, hold down the shift key and it will bring up the boot menu. Then pick the kernel that has “xenomai” in it.

Yogi

apantazis82, 2018-08-21:
Thank you for the reply. Nothing changes when I hold down the Shift key -- I still boot in the same kernel, and RTXI won't start. I only get a single option ("Ubuntu") when I use the boot menu (in my computer that's using the F12 key).

Is there maybe a way to boot directly into xenomai kernel?

yapatel, 2018-08-22:
> Reinstalled from liveCD, this time opting out of online updates -- now it works fine! Maybe the automatic updates messed up the xenomai installation.

Great; I’ll make sure we find a solution for this in the next livecd

Yogi

sudorook, 2018-08-23:
It's possible to specify a default kernel by editing /etc/default/grub. Edit the GRUB_DEFAULT variable to read (I think):
```
GRUB_DEFAULT='Advanced options for Ubuntu>Ubuntu, with Linux 4.9.51-xenomai-3.0.5-aufs'
```
Then, run `update-grub` to rebuild the grub.cfg file.

I think the reason why you are running into kernel issues after running upgrades is that Ubuntu's default non-RT kernel is version 4.15, and the RTXI kernel is 4.9.

When Ubuntu boots, it uses a list of kernels and boot options in /boot/grub/grub.cfg to populate the GRUB menu. The kernels are sorted numerically, so kernel 4.15 (non-RT) comes before 4.9 (RT). The "Ubuntu" option you see at the top of the GRUB boot menu maps directly to the first option in the sorted kernel list.

When you run upgrades, Ubuntu will install its most recent version of kernel 4.15, which is absent on the live CD. The result is that after upgrades are complete, there's a new 4.15 kernel that gets put at the top of the list and gets booted by default, causing you to reboot into a non-RT kernel unawares.

Using the GRUB_DEFAULT method should keep the default kernel mapped to the RT one after you install upgrades and any new non-RT kernel versions that Ubuntu distributes by default.

___

Additionally, I've cobbled together a script that will set a default kernel for you. It looks for RT kernels in your grub.cfg, and if it finds any, it prompts you for the one you wish to set as the default. Enter the number corresponding to the desired kernel, and it will prompt you for your password. Just in case, it also saves a backup of your original /etc/default/grub file inside /etc/default/.

https://github.com/sudorook/rtxi-scripts/blob/master/utils/kernel-picker.sh

closed 2018-11-08 by yapatel