RTXI/rtxi #129: Cannot boot real-time kernel (4.9.24-xenomai-3.0.5)

jadwn, 2019-01-18:
## Basic Information: 
**RTXI Version: 2.2** 
**Installed from Live CD?: No, from source code**

## Error Description: 

**What happened?**
The realtime kernel was compiled successfully. When booting with the 4.9.24-xenomai-3.0.5 kernel, the boot up hangs up at the loading initial ramdisk step. Booting with the generic linux kernel works normally.

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```
Loading initial ramdisk ...
```

yapatel, 2019-01-18:
Have you tried the liveCD?

Yogi

jadwn, 2019-01-18:
I first tried using the liveCD. It seems to be installing the realtime kernel and RTXI properly. However, the analogy drivers are not loaded.
Jad

yapatel, 2019-01-18:
Can you run the liveCD and run the following at a terminal

dmesg > dmesg.out

And send that file to yogi at rtxi.org?

Yogi

yapatel, 2019-01-28:
System worked as expected upon insertion of a compatible DAQ.

closed 2019-01-28 by yapatel