RTXI/rtxi #130: BNC-2090 converter receives an input but does not send an output

eomorozova, 2019-02-08:
Hello,

I am trying to set up a new dynamic clamp system in addition to an existing working one in our lab. I installed RTXI v2.2. I have NI PCIe-6259 DAQ connected to BNC-2090 converter. BNC-2090 receives an input but does not send an output, even though RTXI oscilloscope shows a nonzero output signal. I've noticed that there are 32 inputs and 4 outputs in the control panel of RTXI, while my BNC-2090 converter has only 16 inputs and 2 outputs. I am not sure if my problem has something to do with this.

yapatel, 2019-02-08:
> I've noticed that there are 32 inputs and 4 outputs in the control panel of RTXI, while my BNC-2090 converter has only 16 inputs and 2 outputs.
Yes, if you want to use all I/O on your DAQ you have to use two breakout boards.

Regarding the issue, you should plug up the output of the DAQ to a physical oscilloscope and check what is coming out. If you see nothing on the physical oscilloscope, you will need to put the DAQ into a PC and test the card’s IO functionality with the NI tools. 

Yogi

eomorozova, 2019-03-25:
Hello Yogi,

Thank you for your response. I tested the DAQ with NI MAX software on windows machine and a physical oscilloscope. The output seems to work fine, I did see the output signal on the oscilloscope.
I also tried to change the scaling factor in a System control panel a lot and it does not seem to do anything.

closed 2021-04-01 by fusge