RTXI/rtxi #132: Loading initial ramdisk, problem with PCI-6289

tjcdecoster, 2020-07-21:
## Basic Information: 
**RTXI Version: 2.2** 
**Installed from Live CD?:  No, downloaded with git and compiled from source** 
**Ubuntu version: 16.04 LTS**

## Error Description: 

**What happened?**
The realtime kernel was compiled successfully. When booting with the 4.9.24-xenomai-3.0.5 kernel, the boot up hangs up at the loading initial ramdisk step. Booting with the generic Linux kernel works normally.

**What steps will trigger the error?**
Booting the  PC with the RT-kernel

## Error Message: 

```
Loading initial ramdisk ...
```

## More info
We want to use and install a new set-up in our lab. We bought the PCI-6289 as our preferred DAQ-card since it was specified in the supported DAQ-cards. However, now I am stuck at the "loading initial ramdisk" stage. After googling the error and looking through the solved issues, I found this has to do with the DAQ card that is not compatible. However, since it was mentioned that there should be no problem, I dug a little deeper and found the  following forum post:
https://xenomai.org/pipermail/xenomai/2016-January/035743.html

However, I am not able to get into the link that is mentioned there to look at a possible solution. Could you or anyone else on this forum help me resolve this issue?

I also emailed you the dmesg.out file since you requested it in the other forum post (command was run from the generic Linux version that was installed.)

With kind regards,
Tim

tjcdecoster, 2020-07-23:
# Follow-up for people with the same issue:

## partial solution
I found a way to access that link. The drivers for PCI-6289 are commented out in the pcimio.c. This has to do with the fact that these drivers were still in development when xenomai 3.0.5 was released. However, they are already programmed partially. So, what can you do about this to circumvent that issue:

install with
`./install_rt_kernel.sh`

First the necessary files will be downloaded to your desktop. Afterwards, a screen will open with all the options in there. Now open a different terminal and do:
`cd /opt/xenomai-3.0.5/kernel/drivers/analogy/national_instruments`
`emacs pcimio.c`

When you open the file, you will see that there are two parts within if-conditions. Those you have to uncomment.
Also, upon checking the specifications for the PCI-6289, I noticed that one line should be changed as well:
`ao_speed:	350,`

Does this fix the issue with the ¨loading initial ramdisk¨? No, but it is probably a small step on the way to a solution.

If anyone knows what the 63 options are that are commented in the pcimio.c file, please also let me know. I include the comment for reference:
```
/* These are not all the possible ao ranges for 628x boards.
 They can do OFFSET +- REFERENCE where OFFSET can be
 0V, 5V, APFI<0,1>, or AO<0...3> and RANGE can
 be 10V, 5V, 2V, 1V, APFI<0,1>, AO<0...3>.  That's
 63 different possibilities.  An AO channel
 can not act as it's own OFFSET or REFERENCE.
*/

static struct a4l_rngtab rng_ni_M_628x_ao = { 8, {
	RANGE(-10, 10),
	RANGE(-5, 5),
	RANGE(-2, 2),
	RANGE(-1, 1),
	RANGE(-5, 15),
	RANGE(0, 10),
	RANGE(3, 7),
	RANGE(4, 6),
	RANGE_ext(-1, 1)
}};
static struct a4l_rngdesc range_ni_M_628x_ao =
	RNG_GLOBAL(rng_ni_M_628x_ao);
```

## Remaining problem
I still get the eternal ¨loading initial ramdisk¨ error. However, I think I might have found some posts on the internet that can solve this issue too, although I am not too certain.

Things I tried:

- booting in standard mode: result = ¨loading initial ramdisk¨
- booting in upstart mode: result =  ¨loading initial ramdisk¨
- booting in recovery mode: result = 
```
[   0.058600] x86: Booting SMP configuration:
[   0.058632] .... node  #0, CPUs:      :1
```

So, based on that output, I found the following links:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=886485
https://xenomai.org/pipermail/xenomai/2017-September/037723.html
https://xenomai.org/pipermail/xenomai/2019-March/040462.html

Anybody with any experience in these things?

With kind regards,
Tim

tjcdecoster, 2020-07-23:
# New update (still not solved):

I tried re-installing. The problem seems to be in the linux-headers. One should disable the MAXSMP feature in the .config file. However, whenever the script ./install_rt_kernel runs, the headers get downloaded and installed again.  It does also not seem possible to install the xenomai kernel from scratch to make it compatible with the RTXI-softwarre once all the files are already on your computer.

Anybody who can help?

With kind regards,
Tim

fusge, 2021-04-01:
Do you still have this issue? I will be working on a new version of rtxi that uses xenomai 3.1 and so far i have not encountered this particular problem with drivers.

tjcdecoster, 2021-04-01:
Thank you very much for the answer. That's great news that someone will be producing a new RTXI version!

To answer your question: the issue is still there, but is an issue that is very specific to some cards (628x cards).
Our group now bought different DAQ cards (less output channels and less resolution) that are not part of this line and for which the pcimio.c file is well configured. As a consequence, all is working now.

However, that being said, I would most definitely like to have a working version for our more powerful card, as we had specific applications in mind for this one and currently it is just sitting on a shelf. Is there any information I can provide you with that would be helpful for the development of this new RTXI version? I would be glad to help out (by providing information, but also by small contributions to coding parts if needed).

With kind regards,
Tim

fusge, 2021-04-01:
It's good that you have a working setup. For now it would be best to relase an updated version of rtxi and have you test that version against your most powerful card and then work from there.

tjcdecoster, 2021-04-01:
Ok, let me know when it is available and I'll test it (Is there an approximate release date for example?). Should I note any hardware specifics for the computer itself? With the previous version, you needed to have a processor that was quite old and which took our lab a long time to find.

If you would like to take this conversation away from this forum post, my email can be found on [my university page](https://hartlongcentrum.nl/research/laboratory-of-experimental-cardiology/people/tim-de-coster-phd/)

Also, in case of any other issues/questions I might be of help with, don't hesitate to contact me.

fusge, 2021-07-28:
Hi @tjcdecoster 

A new version of rtxi was recently released (RTXI 2.3) which includes updates to installation scripts and dependencies along with relevant changes to installation steps on the site. Feel free to test it out on your system and contact me if there are any issues.

closed 2022-04-29 by fusge