RTXI/rtxi #133: Integrated graphics card still essential?

sonyadave, 2020-08-11:
Hello,

Is an integrated graphics card still essential? For standard installation?
I don't know if we have one without integrated graphics card.  we can certainly get it. This is my output of lspci:

00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09) (prog-if 00 [VGA controller])
        DeviceName:  Onboard IGD
        Subsystem: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller
        Flags: bus master, fast devsel, latency 0, IRQ 35
        Memory at fe000000 (64-bit, non-prefetchable) [size=4M]
        Memory at c0000000 (64-bit, prefetchable) [size=256M]
        I/O ports at f000 [size=64]
        [virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: i915
        Kernel modules: i915

this is the whole story.  I tried installing the custom RTXI.  I had numerous problems, including the mentioned (on other recent threads)  "loading intitial ramdisk" freeze.

I gave up and went for standard install.  I installed and ran benchmark.
My benchmark times were good.   at the end of benchmark report it said:
RTS|      0.929|      1.586|      8.680|       0|     0|    00:10:00/00:10:00
pcilib: sysfs_read_vpd: read failed: Input/output error
pcilib: sysfs_read_vpd: read failed: Input/output error

when I type rtxi (presumambly how one runs it) i get:

   0"000.000| BUG in low_init(): [main] binding failed: Operation not permitted
thank you,
Sonya

sonyadave, 2020-08-11:
sorry, I mean,is integrated graphic card still "a problem" not "essential"

sonyadave, 2020-08-11:
Also, I have a SSD hard drive.  I saw the RTXI authors wrote something to the effect that hardware modifications in the graphics card to speed it up are non compatible.  SSD is a hard ware modification, but not in graphics card.

Is a solid state drive (SSD) harddrive okay?

sonyadave, 2020-08-12:
Problem seems like it is fixed. I can start the RTXI gui.
I fixed "loading initial ramdisk" by using the standard installation, starting the program with "sudo rtxi" (emphasis on sudo), and holding shift during bootup to get the boot menu.

I also did some updates to ubuntu 16.04.  Not sure if that was essential.  I also did a custom installation. I don't think that was essential, and now my bootup menu shows two versions on xenomai. One runs, the other still shows "loading initial ramdisk".

fusge, 2022-04-29:
Closing as this seems resolved

closed 2022-04-29 by fusge