RTXI/rtxi #139: Error 2 while Building RT kernel on Ubuntu 20.04

lightingnerd, 2021-09-01:
## Basic Information: 
**RTXI Version:** 2.3

**Installed from Live CD?:** I installed ubuntu from a live thumb, and followed the installation instructions on the RTXI website ([rtxi.org/install](rtxi.org/install))

## Error Description: 

**What happened?**
Executed sudo ./install_rt_kernel.sh (had to use sudo to give the script the correct permissions).
The script paused several times to ask if certain items were [N/m/y/?], I guessed to my best ability, and selected "m" for almost all of them. The Kernel Configuration popped up, I went through and set the correct flags. Hit "X" to exit, and selected to save the config. Script ends in "Error 2"

**What was supposed to happen?**
I expected the script to complete and indicate that it was successful in building the rt Kernel.

**What steps will trigger the error?**
1. Install Ubuntu (20.04) or Lubuntu  (20.04) (error occurred on both)
2. Install git
3. Clone rtxi repo
4. Move to /home/ or keep in local user directory (error occurred when trying both)
5. Go into scripts.
6. install dependencies with sudo (the script needed su privileges)
7. install rt kernel with sudo (the script needed su privileges)
8. Randomly guess  [N/m/y/?] to several prompts regarding device drivers
9. Follow the rtxi.org/install guide instructions to set flags in the Kernel Configuration
10. Close with "X" and elect to save, or just hit exit (error occurred after trying both)

## Error Message: 

<--!
 Copy error messages and terminal output between the two braces below.
 --> 

```
make[1]: Entering directory '/opt/linux-cip-4.19.177-cip44'
scripts/kconfig/conf  --syncconfig Kconfig
  DESCEND  objtool
  HOSTCC  scripts/genksyms/genksyms.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/unifdef
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/mod/mk_elfconfig
  YACC    scripts/genksyms/parse.tab.c
  CC      scripts/mod/devicetable-offsets.s
  MKELF   scripts/mod/elfconfig.h
  LEX     scripts/genksyms/lex.lex.c
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/selinux/mdp/mdp
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/conmakehash
  YACC    scripts/genksyms/parse.tab.h
  HOSTCC  scripts/recordmcount
  HOSTCC  scripts/sortextable
  HOSTCC  scripts/genksyms/parse.tab.o
  INSTALL usr/include/asm-generic/ (37 files)
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/asn1_compiler
  INSTALL usr/include/drm/ (26 files)
  HOSTCC  scripts/sign-file
  HOSTCC  scripts/extract-cert
  HOSTCC  scripts/mod/sumversion.o
  INSTALL usr/include/linux/ (499 files)
  HOSTCC  scripts/insert-sys-cert
  INSTALL usr/include/misc/ (2 files)
  HOSTLD  scripts/genksyms/genksyms
  INSTALL usr/include/linux/android/ (1 file)
  INSTALL usr/include/mtd/ (5 files)
  UPD     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  INSTALL usr/include/rdma/ (25 files)
[... 374 lines cut ...]
  CC      arch/x86/kernel/check.o
  CC      arch/x86/kernel/pci-swiotlb.o
  CC      arch/x86/kernel/uprobes.o
  CC      arch/x86/kernel/sysfb.o
  CC      arch/x86/kernel/sysfb_efi.o
  CC      arch/x86/kernel/perf_regs.o
  CC      arch/x86/kernel/tracepoint.o
  CC      arch/x86/kernel/umip.o
  CC      arch/x86/kernel/unwind_frame.o
  CC      arch/x86/kernel/audit_64.o
  CC      arch/x86/kernel/amd_gart_64.o
  CC      arch/x86/kernel/aperture_64.o
  CC      arch/x86/kernel/mmconf-fam10h_64.o
  CC      arch/x86/kernel/vsmp_64.o
  AS      arch/x86/kernel/head_64.o
  CC      arch/x86/kernel/head64.o
  CC      arch/x86/kernel/ebda.o
  CC      arch/x86/kernel/platform-quirks.o
  LDS     arch/x86/kernel/vmlinux.lds
  AR      arch/x86/kernel/built-in.a
  AR      arch/x86/built-in.a
make[1]: Leaving directory '/opt/linux-cip-4.19.177-cip44'
make: *** [debian/ruleset/targets/common.mk:301: debian/stamp/build/kernel] Error 2

```

fusge, 2021-09-02:
Can you run the command `cat /proc/cpuinfo` and paste the contents here? Also What processor type did you select in the kernel configuration?

lightingnerd, 2021-09-04:
I will get back to you with that sometime next week at the latest. 

Thank you!

lightingnerd, 2021-09-11:
Selected the generic x86_64 cpu type in the kernel config.

cat /proc/cpuinfo  gave this:
```processor       : 0
vendor_id       : AuthenticAMD
cpu family      : 21
model           : 2
model name      : AMD FX(tm)-6300 Six-Core Processor
stepping        : 0
microcode       : 0x6000852
cpu MHz         : 1400.000
cache size      : 2048 KB
physical id     : 0
siblings        : 6
core id         : 0
cpu cores       : 3
apicid          : 16
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb cpb hw_pstate ssbd ibpb vmmcall bmi1 arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold
bugs            : fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips        : 7000.41
TLB size        : 1536 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro

processor       : 1
vendor_id       : AuthenticAMD
cpu family      : 21
model           : 2
model name      : AMD FX(tm)-6300 Six-Core Processor
stepping        : 0
microcode       : 0x6000852
cpu MHz         : 1608.635
cache size      : 2048 KB
physical id     : 0
siblings        : 6
core id         : 1
cpu cores       : 3
apicid          : 17
initial apicid  : 1
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb cpb hw_pstate ssbd ibpb vmmcall bmi1 arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold
bugs            : fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips        : 7000.41
TLB size        : 1536 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro

processor       : 2
vendor_id       : AuthenticAMD
cpu family      : 21
model           : 2
model name      : AMD FX(tm)-6300 Six-Core Processor
stepping        : 0
microcode       : 0x6000852
cpu MHz         : 1400.000
cache size      : 2048 KB
physical id     : 0
siblings        : 6
[... 77 lines cut ...]
model name      : AMD FX(tm)-6300 Six-Core Processor
stepping        : 0
microcode       : 0x6000852
cpu MHz         : 1400.000
cache size      : 2048 KB
physical id     : 0
siblings        : 6
core id         : 5
cpu cores       : 3
apicid          : 21
initial apicid  : 5
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb cpb hw_pstate ssbd ibpb vmmcall bmi1 arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold
bugs            : fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips        : 7000.41
TLB size        : 1536 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro
```
Thank you for looking into this!

fusge, 2021-09-15:
I was able to recreate the same error you had by re-patching he kernel a second time on top of an existing installation/patched kernel. A solution would be to remove the following folders:

/opt/xenomai-3.1
/opt/linux-cip-4.19.177-cip44

After doing so re-run the install_rt_kernel.sh script. When the script pauses on device drivers, just answer N. You may have to select again kernel options. Let me know if the problem persists.

lightingnerd, 2021-09-15:
That appears to be the issue, it has installed successfully now. The first time, the power to the building had shut off mid-install, and during both the first and the second time, the build process was interrupted because my console window was too small to access the kernel configurations options. 

By removing those two directories, and ensuring the console window was large enough at the beginning, the RT kernel built and installed successfully, and the rtxi install script just completed successfully as well. 

Thank you!

closed 2021-09-15 by lightingnerd