RTXI/rtxi #158: Connector will not add connections that close the loop

thiagoburghi, 2024-04-09:
## Basic Information: 
RTXI Version: 3.0.3
Installed from Live CD?: No (compiled from source)
DAQ device: NI PCIe-6259

## Error Description: 
In the Connector panel:
After connecting an output of a widget to an input of the DAQ, it becomes impossible to connect an output of the DAQ to an input of the same widget.
Conversely, after connecting an output of the DAQ to an input of a widget, it becomes impossible to connect an output of the same widget to an input of the DAQ.

What happened?
Clicking the "Connect" button will not add the connection that would have closed the loop between DAQ and widget.

What was supposed to happen?
The connection should be added.

What steps will trigger the error?
The error seems to happen with any widget, but it can be tested with the FIR filter widge that is readily available after installation: 
1) Open RTXI
2) load the fir-window widget
3) open the Connector widget
4) Connect output of fir-window to input of the DAQ: connection is added
5) Attempt to connect output of the DAQ to fir-window: connection is not added 

## Terminal message:
(does not seem to indicate there is an error)
```
CONNECTION -- {source: fir-window type: Output port: 0} <==> {destination: PCIe-6259 port: 0}
[ 2024-04-09 05:40:25 ] (TELEMITRY)	 TYPE -- IO Link Updated
[ 2024-04-09 05:40:25 ] (EVENT FIRED)	 TYPE -- CONNECTOR : Block list requested
[ 2024-04-09 05:40:25 ] (EVENT FIRED)	 TYPE -- CONNECTOR : Request for all connections
[ 2024-04-09 05:40:25 ] (EVENT FIRED)	 TYPE -- CONNECTOR : Block list requested
[ 2024-04-09 05:40:25 ] (EVENT FIRED)	 TYPE -- CONNECTOR : Request for all connections

```

fusge, 2024-04-09:
Thanks for submitting this. Seems to be an issue with the connector not recognizing that DAQ devices are allowed to form cycles in the connection graph. Should be an easy fix.

thiagoburghi, 2024-04-09:
Thanks for looking into this so quickly! We did a quick test and it's still not allowing us to close the loop between widget and DAQ device. The DAQ device is now able to form a closed loop with itself, though. It looks like the problem is that when connecting the widget to the DAQ, the check of whether there is a closed-loop is still the same, and since it starts with a non-DAQ component it won't allow the loop to be formed.

fusge, 2024-04-10:
Added a better fix. Will check very soon to confirm it properly connects on our devices as well

hxywang7, 2024-04-10:
Thanks for such quick fix! Now the connector panel shows the widget can be connected to DAQ device in a closed loop. However it seems like the fix introduces bugs in features dealing with data processing.

We first connected the PCIe output (ai7(, which is just receiving a square wave we generated in a physical waveform generator, to the input of a mimic-signal widget (cloned from Github). Then we connected the output of the mimic-signal widget to an input of the DAQ (ao0):
[image]

When we attempt to see the signal in the Oscilloscope, we don`t see our square wave (we see a zero signal instead)..
[image]

In addition, trying to record the data from the output of the mimic-signal doesn't work (the data file comes out empty).

This seems to be a bug because PCIe output is indeed a square wave: adding the PCIe output to Oscilloscope will make mimic-signal input disappear (it was previously zero), and show the actual signal at the output of the PCIe device: 
[image]

We are wondering now if the closed-loop check is being done elsewhere in the code. Would there be a quick way to disable it?

fusge, 2024-04-12:
Seems to be an issue introduced by modifying the cycle detection logic. A fix is underway and should be out later today.

thiagoburghi, 2024-04-16:
This is now working fine, thanks!

PR #159 (2024-04-09): 158 connector will not add connections that close the loop
RTXI does not allow cycles to form in the connections as this would result in weird and possibly wrong behavior. However, it is, and should be, allowed for connecting DAQ devices with other plugins and devices. This fixes a bug where this was not allowed.

closes #158
PR #162 (2024-04-10): Check both source and destination blocks for devices when doing cycle…
this further fixes #158
PR #163 (2024-04-12): Bugfix/oscilloscope scale and offset
fixes #158, #161, and #160
closed 2024-04-16 by fusge