RTXI/rtxi #160: Oscilloscope Apply button bug

hxywang7, 2024-04-09:
## Basic Information: 
**RTXI Version:** 
3.0.3
**Installed from Live CD?:**
No

## Error Description: 

**What happened?**
In Oscilloscope plugin, when you choose Daq card output channel, for example output channel 3, and click one "enable channel", then click on "apply", it will come back to channel 0 
[image]
[image]
[image]

**What was supposed to happen?**
It should keep same with channel 3

**What steps will trigger the error?**

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```

```

fusge, 2024-04-10:
The number on the label is in fact the ID number that identifies the block being measured, so it will always be 0 in your case if you add additional channels. However there should be a label that identifies the channel being measured as well for clarity. 

I solution would be to provide a more descriptive label that will output both the ID and the channel name. We can certainly add that.

thiagoburghi, 2024-04-10:
Hi Ivan,

I think Harry meant that the channel label (ai3 and ai0) are the ones that
are changing after the button is clicked. I think they are already the
labels that identify the channels.
If I may add to Harry's request, the "input/output" labels in the
oscilloscope seem to be reversed with regards to those of the control
panel.

Best wishes,

Thiago

fusge, 2024-04-12:
> I believe Harry meant that the channel label (ai3 and ai0) are the ones that
are changing after the button is clicked. I think they are already the
labels that identify the channels.

Yes I realized after I made the comment which is why it was deleted to avoid confusion. Sorry about that.

>If I may add to Harry's request, the "input/output" labels in the
oscilloscope seem to be reversed with regards to those of the control
panel.

The reason why the oscilloscope behaves that way is because a physical input into the DAQ device would appear from RTXI's perspective as an output from a plugin that should be pushed to other plugins. Conversely writing to a DAQ plugin input would result in a physical output by the device. This is only relevant for DAQs, as normal plugins such as the mimic-signal would not have anything to do with a real device so the input/output flags match the names.

thiagoburghi, 2024-04-15:
The new version solved it, thanks!

closed 2024-04-15 by fusge