RTXI/rtxi #165: Data Recorder does not record from multiple DAQ channels

thiagoburghi, 2024-04-15:
## Basic Information: 
RTXI Version: 3.0.4
Installed from Live CD?: No
DAQ model: NI PCIe-6259

## Error Description: 
What happened?
Selecting more than one channel of the DAQ to record in the Data recorder leads to only one of the channels being recorded.

What was supposed to happen?
All the selected channels should be recorded.

What steps will trigger the error?
- Open control panel and activate channels that should be recorded; click "apply"
- Open the Data Recorder widget
- Select the DAQ in the "Block" drop-down menu 
- Select one channel in the "Channel" drop-down menu, and click "add"
- Select other channels in the "Channel" drop-down menu, and click "add"
- Choose file name and start recording
- Stop recording and check the content of the .h5 file (e.g., using the data analysis widget but any other means will yield same result)
- Verify that only one channel of the DAQ has been recorded 

PS: The "Currently Recording" window of the Data Recorder shows the name of the DAQ for all the selected channels (it does not specify which channel that is). So it could simply be a problem of naming the data accordingly.

fusge, 2024-04-15:
A fix has been applied. It originated from NIDAQ driver incorrectly filling device names. The fix is already available in the dev branch.

thiagoburghi, 2024-04-16:
Tested this today and it is working, thanks!

closed 2024-04-16 by fusge