RTXI/rtxi #166: Channels get scrambled when activating new channels

thiagoburghi, 2024-04-16:
## Basic Information: 
RTXI Version: 3.0.4
Installed from Live CD?: No
DAQ device: NI PCIe 6259

Error Description: Activating a new DAQ channel in the control panel changes the signal of previously activated channels

What was supposed to happen: signals should remain associated to their correct channels

What steps will trigger the error?
1- Turn on Oscilloscope to monitor signals.
2 - Activate a channel (in the control panel) that is currently receiving a signal and display in oscilloscope. 
Example: channel 3 of our DAQ is just receiving white noise:
[image]

3- Activate a second channel (in the control panel) that is currently receiving a different signal. The previously activated channel now displays the different signal. Example: after activating channel 6, channel 3 now display the different signal (pink noise):
[image]

In addition, channel 6 now displays the signal that was previously associated to channel 3:
[image]

4-Activating new signals keep scrambling the previous channels. Example: activated channel 7, now channel 6 is yet a different signal:
[image]

While it is now channel 7 that acquired the white noise signal that should have been that of channel 3:
[image]

5 - **Deactivating the last activated channel seems to correct everything**. Example: deactivating channel 7 results in channels 3 and 6 having the right signals:
[image]

The problem is not one with the oscilloscope: recording the signals shows that the data associated to them has indeed been scrambled internally in RTXI. Here is data recorded after step 4:
[image]

fusge, 2024-04-17:
The error comes from the National Instruments DAQ driver for RTXI. It reads and writes channels in the order the channels are added instead of a "sorted by port number" order. The issue is fixed and should be available in the dev branch soon.

fusge, 2024-04-17:
I merged the fix to the dev branch. Can you confirm that the update fixes the issue on your hardware?

thiagoburghi, 2024-04-19:
I can confirm that the fix worked, thanks!

PR #167 (2024-04-17): Simplified logic for task configuration when enabling/disabling DAQ c…
fixes #166
closed 2024-04-19 by fusge