RTXI/rtxi #173: Initial startup: Unable to create real-time buffer

harrison-e, 2024-07-10:
## Basic Information: 
**RTXI Version:** 3.0.4 
**Installed from Live CD?:** No, compiled from source

## Error Description: 

**What happened?** 

I freshly installed Ubuntu 22.04.4 from USB, and immediately followed the instructions on rtxi.org to install Xenomai and RTXI. After rebooting in the Xenomai kernel, running `./scripts/install_rtxi.sh` with the `evl` option, and going to run `sudo rtxi`, I was met with an error (See Error Message).

**Some extra info:**

After some digging, I see that this error is (probably) coming from a function in `fifo_evl.cpp`, in the function starting on line 48. It seems that evl is returning a 0 or negative value for `xbuf_fd`, but the evl core error code is nondescriptive ("Unknown error -2"). I'm very unfamiliar with both the RTXI and evl codebase, and I can't find a correspondence between the error codes in evl and any description online ([I'm looking at the docs on the Xenomai website](https://evlproject.org/core/kernel-api/xbuf/)).

If you have any insight into what might be happening here with Xenomai, it would be greatly appreciated!!

**What was supposed to happen?** 

RTXI was supposed to open for the first time.

**What steps will trigger the error?**

`sudo rtxi`

## Error Message: 

Copy error messages and terminal output between the two braces below. 

```
user@MACHINE:/home/rtxi$ sudo rtxi
Welcome to RTXI Version 3.0.4
RT::OS::FIFO(evl) : Unable to create real-time buffer

evl core : Unknown error -2
RT::OS::getFifo(EVL) : No such file or directory
RT::System::System : failed to create Fifo
signal_handler : signal type Segmentation fault received

Backtrace:

/lib/librtxi.so(_Z15PRINT_BACKTRACEv+0x48)[0x7f49fa83bcb8]
rtxi(+0x28f82)[0x558a417f3f82]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7f49f8fd6520]
/lib/librtxi.so(+0x24fd5)[0x7f49fa842fd5]
/lib/x86_64-linux-gnu/libstdc++.so.6(+0xdc253)[0x7f49f93a0253]
/lib/x86_64-linux-gnu/libc.so.6(+0x94ac3)[0x7f49f9028ac3]
/lib/x86_64-linux-gnu/libc.so.6(+0x126850)[0x7f49f90ba850]
QMutex: destroying locked mutex
QMutex: destroying locked mutex
QMutex: destroying locked mutex
```

*Thank you for reading!! I see there's only one active maintainer on the RTXI website, so if this is you, thank you for your service!!*

harrison-e, 2024-07-11:
**Update:** 

Issue was resolved by reinstalling Xenomai kernel as root, and changing Processor type in kernel config. Thanks!

closed 2024-07-11 by harrison-e

fusge, 2024-07-13:
Just curious what was the previous and new setting for the processor type in your kernel configuration?

harrison-e, 2024-07-13:
Previous was generic x86 and new is Core 2 / newer Xeon. The CPU used in the machine is Intel Core i9-13900KS.