RTXI/rtxi #177: Crashing when loading workspace

thiagoburghi, 2024-11-13:
## Basic Information: 
**RTXI Version:** 3.1.0

## Error Description: Saving and loading a workspace with control panel open crashes the system.
**What happened?**
RTXI crashes when I attempt to load a workspace that was saved with the control panel open. Saving and loading a workspace without the control panel open seems to work fine.

**What was supposed to happen?**
I should have been able to load the saved workspace without problems.

**What steps will trigger the error?**
Save a workspace with control panel open; close RTXI. Open RTXI, try to load workspace, RTXI crashes.

## Error Message: 

Plugin::load : failed to retrieve getFactories symbol
signal_handler : signal type Segmentation fault received

Backtrace:

/lib/librtxi.so(_Z15PRINT_BACKTRACEv+0x48)[0x7f2938b01ea8]
rtxi(+0x2f6c2)[0x56157b0206c2]
/usr/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7f293729b520]
/lib/librtxi.so(_ZNK7Widgets6Plugin26getComponentParametersInfoEv+0x12)[0x7f2938b10872]
/lib/librtxi.so(_ZN7Widgets6Plugin21loadParameterSettingsER9QSettings+0x65)[0x7f2938b13315]
rtxi(+0x2e7d4)[0x56157b01f7d4]
rtxi(+0x2788f)[0x56157b01888f]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f1793)[0x7f2937ad0793]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN7QAction9triggeredEb+0x46)[0x7f2938581be6]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN7QAction8activateENS_11ActionEventE+0xac)[0x7f29385848fc]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2f3142)[0x7f293870f142]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2fad39)[0x7f2938716d39]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN7QWidget5eventEP6QEvent+0x20e)[0x7f29385cb4ee]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x83)[0x7f2938588713]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0xa84)[0x7f2938590364]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x13a)[0x7f2937a98e3a]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Ebb+0x1b7)[0x7f293858ee47]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c9843)[0x7f29385e5843]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5)[0x7f29385e7fd5]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x83)[0x7f2938588713]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x13a)[0x7f2937a98e3a]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN22QGuiApplicationPrivate17processMouseEventEPN29QWindowSystemInterfacePrivate10MouseEventE+0x697)[0x7f2937e7f307]
/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5(_ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0xac)[0x7f2937e54a2c]
/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e)[0x7f293126ed6e]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b)[0x7f2936303d3b]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8)[0x7f29363592b8]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33)[0x7f29363013e3]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x68)[0x7f2937af20b8]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x12b)[0x7f2937a9775b]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication4execEv+0x94)[0x7f2937a9fcf4]
rtxi(+0x21613)[0x56157b012613]
/usr/lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x7f2937282d90]
/usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80)[0x7f2937282e40]
rtxi(+0x21ac5)[0x56157b012ac5]
signal_handler : signal type Segmentation fault received

Backtrace:

/lib/librtxi.so(_Z15PRINT_BACKTRACEv+0x48)[0x7f2938b01ea8]
rtxi(+0x2f6c2)[0x56157b0206c2]
/usr/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7f293729b520]
[0x56157c3c5a30]

Yunusbcr, 2025-02-24:
@ #181

fusge, 2025-03-02:
Hey I've added new changes that clarifies things on the connector so that it does load connections correctly. Please checkout the latest changes on the dev branch and test on your end with devices present and let me know if this fixes your issue.