RTXI/rtxi #182: RTXI Not Running in Real-Time Mode Despite EVL Kernel Installation

hxywang7, 2025-03-16:
When launching RTXI, I receive the warning:
> You are using the POSIX compatibility layer, RTXI is NOT running in realtime!!

I have installed RTXI using the EVL kernel, and my system reports running on 5.15.129-xenomai-4 when using `uname -r`
However, after verifying with the Xenomai testing suite (https://v4.xenomai.org/core/testing/), all tests pass, indicating that I should have a functioning EVL kernel core.

Despite this, RTXI does not seem to recognize the real-time environment.

System Information
-   OS: Ubuntu 22.04.5 LTS
-   Kernel Version: 5.15.129-xenomai-4
-   RTXI Version: 3.2.0

  

[image]

fusge, 2025-03-19:
It may be that the wrong version was installed, or that the installation occurred while on the non-realtime kernel. Either way you have to reinstall RTXI.

Please uninstall RTXI using the following commands. Before doing so make sure you have any data and settings backed up:

```sh
dpgk -r rtxi
dpgk -p rtxi
```

The above commands remove and purge all RTXI related files and plugins. Restart your computer Make sure that you are booted in the real-time kernel. Before continuing, If using National Instruments then also make sure that the support libraries and drivers are also installed or RTXI may not build with the needed drivers to interface with them. 

Next Follow the instructions for installing RTXI. Select the EVL option for the Real-Time Kernel.

If the problem persists please respond here and we'll continue from there.

fusge, 2025-03-21:
It may be that the build system can't find the evl library. Can you post the output of the following command?

```sh
which evl
```

If it does find a binary, please run the following command as well:

```sh
evl test
```

hxywang7, 2025-03-24:
Thanks for your response!

Here's the output of the commands you suggested:
```
samsep10l@samsep10l-MS-7D78:~$ which evl
/usr/local/bin/evl
```
```
samsep10l@samsep10l-MS-7D78:~$ evl test
evl: no 'test' command. Try 'evl --help'.

```
Looks like the evl binary is found, but test isn't a valid command. Let me know what else I can check!

fusge, 2025-03-29:
That's odd. To double check please also run the following command:

```bash
pkg-config --list-all | grep evl
```

What output do you see?

hxywang7, 2025-03-31:
## ✅ Issue Solved

Thanks for the support! The problem has been solved.

After investigation, the root cause turned out to be a **stale `librtxipal.so`** installed manually in `/lib/` from a previous build. This file was not managed by `dpkg` or CMake's packaging system.

Although `install_rtxi.sh` correctly built and installed the EVL version of `librtxipal.so` into `/usr/local/lib`, RTXI was actually **loading the outdated library from `/lib`** due to it appearing earlier in the dynamic linker search path.

---

### ✔️ Solution steps:

1. Removed all stale copies:
    ```bash
    sudo rm -f /usr/lib/librtxipal.so
    sudo rm -f /usr/local/lib/librtxipal.so
    sudo rm -f /lib/librtxipal.so  # <-- this was the hidden culprit
    ```

2. Rebuilt RTXI cleanly and Installed the correct version:
    ```bash
     rm -rf build
    ./scripts/install_rtxi.sh
    ```

4. Verified that RTXI now correctly uses EVL:
    ```
    RTXI: Successfully attached thread to EVL!
    ```

---

### 💡 Recommendation:
It might be helpful to add a check in `install_rtxi.sh` to detect old `librtxipal.so` files in `/lib`, as they can cause confusion and are not managed by the package manager.

Thanks again for your help!